motioneye-project/motioneye #1972: Plugin platform.

surak, 2020-12-20:
Hi,

recently I wanted to do something more with the images collected by my camera, and I realized that there is no plugin platform on MotionEye.

While 'shopping' around for a solution, I stumbled upon pi-timolo [1], a software which does part of what motioneye does, but in a way less sophisticated way.

Turns out that there is a rudimentary, yet effective way to use plugins on it: a python function where you add your code, and gets called when the main code saves a file [2]. It's just a `def userMotionCode(filenamePath):`, which allows one to add code to perform operations on saved files. In my case, I add labels to them. 

I would rather keep using motioneye and motioneyeos instead, as the stability has been nothing short of legendary. I have cameras where the only downtime is from upgrades.

If such a thing exists already, please let me know. If not, here is my suggestion. If you don't intend to do such a thing, just let me know, too. Thank you. 

[1] https://github.com/pageauc/pi-timolo/tree/master/source
[2] https://github.com/pageauc/pi-timolo/blob/f95b54916c528a44bb5cda75f3011fa86f412816/source/user_motion_code.py#L18

starbasessd, 2020-12-20:
If you have Motion Detection on, you should see Motion Notificatons just
below.
There are several options there, Run a Web Hook, Run a Command, and Run an
end command.
You can place you stuff in the /data folder tree and call it.
This works for both motionEye and motionEyeOS.
You cannot add extra python functions or applications to motionEyeOS
because it's locked down, but it's possible to do some functions.
If you want to do major add-ons, I'd recommend RaspberryPiOS Lite, and
installing motion/motionEye from the instructions found here:
Home · ccrisan/motioneye Wiki <https://github.com/ccrisan/motioneye/wiki>

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

surak, 2020-12-20:
@starbasessd thanks for the reply! Now that you mentioned, that is exactly what I want, with two caveats:

- For this one, one should need to figure which file was added and where - this could easily be a system environment variable for example (so it would work in any programming language), so any script could know which file was created.
While the web hook allows one to use some variables, that does not seem to be the case with the "run a command" thing.
- As you mentioned, it's locked down. An ideal case would be to be able to upload a code via the web interface, or a place where you could do something, like the given file example from the other project.

What do you think?

starbasessd, 2020-12-20:
As long as your needs do not include needing to use apt, or similar.
Can you give me a 'for example' for the variables needed? You can use an
export command to give a variable to a called program/script.
You can upload via scp/ssh almost anything into the /data folder tree.
Or, you can use another OS like RaspberryPiOS, Ubuntu, whatever that isn't
locked down like motionEyeOS, where you have apt, pip, etc, and free use of
the root folder tree.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

surak, 2020-12-20:
The filename of the file which has been just created is the example. One could traverse the directory tree every time, but as you can see from Motioneye's own "picture browser", that can get slow.

starbasessd, 2020-12-20:
I believe you'd be better off using motion/motionEye on RaspberryPiOS or
Ubuntu on Pi, as a lot of the paths are 'hard-coded' in those 2 examples.
You'd need to change everything that uses /usr folder tree (which is locked
down) and every other hard coded non-/data/ path.
You entered the issue in the motionEye issues, are you trying to use
motionEye or motionEyeOS?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

surak, 2020-12-20:
I use both, depending on the case, to be honest. But I will stick to raspbian with motioneye, then.

starbasessd, 2021-03-05:
If you are still having issues, please open a new issue.

closed 2021-03-05 by starbasessd