motioneye-project/motioneye #1973: [Question] https://github.com/ccrisan/motioneye/wiki/Install-On-Raspbian

ovz93br43v7, 2020-12-21:
I have a question regarding the install instructions on [https://github.com/ccrisan/motioneye/wiki/Install-On-Raspbian](https://github.com/ccrisan/motioneye/wiki/Install-On-Raspbian)

In step 2 you install motion with:
```
wget https://github.com/Motion-Project/motion/releases/download/release-4.2.2/pi_buster_motion_4.2.2-1_armhf.deb
dpkg -i pi_buster_motion_4.2.2-1_armhf.deb
```
The newest release on [https://github.com/Motion-Project/motion/releases](https://github.com/Motion-Project/motion/releases) is 4.3.2. 

Should I use the newest version or take 4.2.2-1 as stated in the instructions?

starbasessd, 2020-12-21:
You can use either, 4.3.2 has shown to have some issues.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

ovz93br43v7, 2020-12-21:
Ok thanks for your fast answer. Then I will use 4.2.2-1.

Can I somewhere check how much power my motioneye server need per vga/720p/1080p camera stream?
Or will my raspberry pi 4 (4 GB) server [current load only ~5%] too weak to handle four fast network camera streams?

starbasessd, 2020-12-21:
I have 4 network cameras on my Pi3s and Pi4s when checking user issues (3
D-Link DCS-5020Ls and a FosCam 9900 RTSP mode) and have few issues. They
won't handle 4 4k cameras at any reasonable framerate, I run around 800x600
(Foscam is at 1280x720) and I average 4-6 fps. I don't expect 60 fps, HD
video, I only paid $35 for the Pis.
My Production motionEye server hub is i5 processor TFF Lenovo and averages
20fps with 5 cameras (the same 4 as above plus a USB Generic Dashcam), but
cost quite a bit more than the Pis.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

closed 2020-12-21 by ovz93br43v7

ovz93br43v7, 2020-12-21:
@starbasessd Sorry but I have to ask another question:
I tried to install on my raspberry pi 4 the packages described here: [https://github.com/ccrisan/motioneye/wiki/Install-On-Raspbian](https://github.com/ccrisan/motioneye/wiki/Install-On-Raspbian)
But only the package `ffmpeg` tries to install 141 additional packages:

```
  ffmpeg fontconfig libaacs0 libaom0 libass9 libasyncns0 libavc1394-0 libavcodec58 libavdevice58 libavfilter7 libavformat58
  libavresample4 libavutil56 libbdplus0 libbluray2 libbs2b0 libcaca0 libcairo2 libcdio-cdda2 libcdio-paranoia2 libcdio18
  libchromaprint1 libcodec2-0.8.1 libcroco3 libdatrie1 libdc1394-22 libdrm-amdgpu1 libdrm-common libdrm-nouveau2 libdrm-radeon1
  libdrm2 libfftw3-double3 libflac8 libflite1 libgdk-pixbuf2.0-0 libgdk-pixbuf2.0-bin libgdk-pixbuf2.0-common libgl1 libgl1-mesa-dri
  libglapi-mesa libglvnd0 libglx-mesa0 libglx0 libgme0 libgraphite2-3 libgsm1 libharfbuzz0b libice6 libiec61883-0 libjack-jackd2-0
  liblilv-0-0 libllvm9 libmp3lame0 libmpg123-0 libmysofa0 libnorm1 libogg0 libopenal-data libopenal1 libopenjp2-7 libopenmpt0 libopus0
  libpango-1.0-0 libpangocairo-1.0-0 libpangoft2-1.0-0 libpgm-5.2-0 libpixman-1-0 libpostproc55 libpulse0 libraw1394-11 librsvg2-2
  librsvg2-common librubberband2 libsdl2-2.0-0 libsensors-config libsensors5 libserd-0-0 libshine3 libsm6 libsnappy1v5 libsndfile1
  libsndio7.0 libsord-0-0 libsoxr0 libspeex1 libsratom-0-0 libssh-gcrypt-4 libswresample3 libswscale5 libthai-data libthai0 libtheora0
  libtwolame0 libva-drm2 libva-x11-2 libva2 libvdpau-va-gl1 libvdpau1 libvidstab1.1 libvorbis0a libvorbisenc2 libvorbisfile3 libvpx5
  libwavpack1 libwayland-client0 libwayland-cursor0 libwayland-egl1 libwebpmux3 libx11-xcb1 libx264-155 libx265-165 libxcb-dri2-0
  libxcb-dri3-0 libxcb-glx0 libxcb-present0 libxcb-render0 libxcb-shape0 libxcb-shm0 libxcb-sync1 libxcb-xfixes0 libxcursor1
  libxdamage1 libxfixes3 libxi6 libxinerama1 libxkbcommon0 libxrandr2 libxrender1 libxshmfence1 libxss1 libxtst6 libxv1 libxvidcore4
  libxxf86vm1 libzmq5 libzvbi-common libzvbi0 mesa-va-drivers mesa-vdpau-drivers va-driver-all vdpau-driver-all x11-common
```

I'm a bit confused whether this is all needed....

starbasessd, 2020-12-21:
Yep, it does.
If you run into errors around the step 3 (python-pip) you can use the
instructions for Ubuntu20.04 or Newer. All the flavors are killing
python2.7 upstream, I'm waiting for them to do the same in RaspberryPiOS

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

ovz93br43v7, 2020-12-27:
@starbasessd I have successfully installed everything. Thank you for your help! Camera's I will link within the next days. One question I have: I have created a "Surveillance User" on my server. But when I try to login the login page is loaded again. Any idea why?

starbasessd, 2020-12-27:
Cached page? Did you set "Remember Me"?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

ovz93br43v7, 2020-12-27:
No "remember me" is not activated. I already tried to use a very simple password, but behaviour is always:
1. I enter username and password
2. Login screen disappears. 
3. Login screen comes back. (no failure message)

When I try to login as admin and I enter a wrong password, I become the message "Invalid credentials.".

Can it be that I have to link first a camera to the server and then the user is active?

starbasessd, 2020-12-27:
Probably not. There have been times when rebuilding a motionEyeOS and I had
saved user and password (Remember Me) and it tries to log in automagically.
I then have to reset everything.
Possible the image is corrupt.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

ovz93br43v7, 2020-12-27:
This installation is not the motioneyeos image. It is the server install according: [https://github.com/ccrisan/motioneye/wiki/Install-On-Raspbian](https://github.com/ccrisan/motioneye/wiki/Install-On-Raspbian)

starbasessd, 2020-12-27:
What is your base OS? (Version?)
Can you run back through the install for Ubuntu 20.04, please?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

ovz93br43v7, 2020-12-27:
OS Version--> Rasbian 10 (buster) armv71
I tested the install instructions of ubuntu 20.04, this way rasbian would install many packages during step 4:

1. `libffi-dev libzbar-dev libzbar0` plus 107 additional packages o.O ->
```autoconf automake autotools-dev bzip2-doc fonts-droid-fallback fonts-noto-mono ghostscript gir1.2-freedesktop gir1.2-gdkpixbuf-2.0
  gir1.2-rsvg-2.0 gsfonts icu-devtools imagemagick-6-common libavahi-client3 libblkid-dev libbz2-dev libcairo-gobject2
  libcairo-script-interpreter2 libcairo2-dev libcups2 libcupsfilters1 libcupsimage2 libde265-0 libdjvulibre-dev libdjvulibre-text
  libdjvulibre21 libexif-dev libexif-doc libexif12 libfontconfig1-dev libgdk-pixbuf2.0-dev libglib2.0-bin libglib2.0-dev
  libglib2.0-dev-bin libgs9 libgs9-common libheif1 libice-dev libicu-dev libijs-0.35 libilmbase-dev libilmbase23 libjbig-dev
  libjbig2dec0 libjxr-tools libjxr0 liblcms2-2 liblcms2-dev liblqr-1-0 liblqr-1-0-dev libltdl-dev liblzma-dev libmagick++-6-headers
  libmagick++-6.q16-8 libmagick++-6.q16-dev libmagick++-dev libmagickcore-6-arch-config libmagickcore-6-headers libmagickcore-6.q16-6
  libmagickcore-6.q16-6-extra libmagickcore-6.q16-dev libmagickwand-6-headers libmagickwand-6.q16-6 libmagickwand-6.q16-dev
  libmount-dev libopenexr-dev libopenexr23 libopenjp2-7-dev libpaper-utils libpaper1 libpcre16-3 libpcre3-dev libpcre32-3
  libpcrecpp0v5 libpixman-1-dev libpthread-stubs0-dev librsvg2-dev libselinux1-dev libsepol1-dev libsigsegv2 libsm-dev libtiff-dev
  libtiffxx5 libtool libwmf-dev libwmf0.2-7 libx11-dev libxau-dev libxcb-render0-dev libxcb-shm0-dev libxcb1-dev libxdmcp-dev
  libxext-dev libxml2-dev libxrender-dev libxt-dev libxt6 m4 poppler-data python3-distutils python3-lib2to3 uuid-dev x11proto-core-dev
  x11proto-dev x11proto-xext-dev xorg-sgml-doctools xtrans-dev
```

2. `python-pil libimagequant0 liblcms2-2 libwebpdemux2 python-olefile`

starbasessd, 2020-12-27:
>    1. python-pil libimagequant0 liblcms2-2 libwebpdemux2 python-olefile

Yep, and another installs 116 packages. Mostly depends for ffmpeg and image
manipulation...

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

ovz93br43v7, 2020-12-28:
I made some trials. After I linked the first camera to my server the login of my server-user worked!

Now the only question is do I really need the additional packages which are installed when I go through the install instructions for ubuntu 20.04 on my rasbian? Or do I get any benefit like improved performance...? :-/

starbasessd, 2020-12-28:
Almost everything is functionality based. Everything you see installed
specifically, is mandatory. All the other files (like the 106 & 115) are so
that the installed packages work. If you want to minimise, I might suggest
using the Docker, or manually compiling.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

ovz93br43v7, 2020-12-28:
No, sorry that wasn't what I wanted to express. It is just:
- Rasbian install instructions steps installed x packages and was done
- Ubuntu install instructions on my Rasbian would install y additional packages

When the y additional packages from the ubuntu way are functional relevant why are they not installed during the rasbian install instructions?