motioneye-project/motioneye #1974: Regular restarts related to failed SMB unmounting

dannytech, 2020-12-22:
My motionEye instance is installed on an Ubuntu 20.04 LXC container running on Proxmox. The motionEye version is 0.42.1.

Every 5 minutes, motionEye apparently restarts, causing a brief disruption in the feed and a weird SMB mounting issue. To be honest, I'm not quite sure what's causing the restart or whether it's the start or the end of one, but here is what the logs look like (result of `journalctl -u motioneye`):

```
Dec 22 14:33:48 tuvok meyectl[21104]: umount: /media/[redacted]: no mount point specified.
Dec 22 14:33:48 tuvok meyectl[1531]:    ERROR: failed to unmount smb share "//[redacted]/surveillance" from "/media/[redacted]"
Dec 22 14:33:50 tuvok motion[21159]: [0:motion] [NTC] [ALL] conf_load: Processing thread 0 - config file /etc/motioneye/motion.conf
Dec 22 14:33:50 tuvok motion[21159]: [0:motion] [NTC] [ALL] config_camera: Processing camera config file camera-1.conf
Dec 22 14:33:50 tuvok motion[21159]: [0:motion] [NTC] [ALL] motion_startup: Logging to syslog
Dec 22 14:33:50 tuvok motion[21159]: [0:motion] [NTC] [ALL] motion_startup: Motion 4.2.2 Started
Dec 22 14:33:50 tuvok motion[21159]: [0:motion] [NTC] [ALL] motion_startup: Using default log type (ALL)
Dec 22 14:33:50 tuvok motion[21159]: [0:motion] [NTC] [ALL] motion_startup: Using log type (ALL) log level (WRN)
Dec 22 14:34:13 tuvok motion[21159]: [1:ml1:Kes] [ERR] [ALL] motion_init: Error capturing first image
Dec 22 14:38:52 tuvok meyectl[21650]: mount error(16): Device or resource busy
Dec 22 14:38:52 tuvok meyectl[21650]: Refer to the mount.cifs(8) manual page (e.g. man mount.cifs) and kernel log messages (dmesg)
Dec 22 14:38:52 tuvok meyectl[21652]: mount error(2): No such file or directory
Dec 22 14:38:52 tuvok meyectl[21652]: Refer to the mount.cifs(8) manual page (e.g. man mount.cifs) and kernel log messages (dmesg)
Dec 22 14:38:52 tuvok meyectl[21656]: umount: /media/[redacted]: no mount point specified.
Dec 22 14:38:52 tuvok meyectl[1531]:    ERROR: failed to unmount smb share "//[redacted]/surveillance" from "/media/[redacted]"
Dec 22 14:38:54 tuvok motion[21711]: [0:motion] [NTC] [ALL] conf_load: Processing thread 0 - config file /etc/motioneye/motion.conf
Dec 22 14:38:54 tuvok motion[21711]: [0:motion] [NTC] [ALL] config_camera: Processing camera config file camera-1.conf
Dec 22 14:38:54 tuvok motion[21711]: [0:motion] [NTC] [ALL] motion_startup: Logging to syslog
Dec 22 14:38:54 tuvok motion[21711]: [0:motion] [NTC] [ALL] motion_startup: Motion 4.2.2 Started
Dec 22 14:38:54 tuvok motion[21711]: [0:motion] [NTC] [ALL] motion_startup: Using default log type (ALL)
Dec 22 14:38:54 tuvok motion[21711]: [0:motion] [NTC] [ALL] motion_startup: Using log type (ALL) log level (WRN)
Dec 22 14:39:17 tuvok motion[21711]: [1:ml1:Kes] [ERR] [ALL] motion_init: Error capturing first image
```

I have no idea why, but for some reason it keeps trying to unmount the share, and failing for weird reasons I can't explain. The SMB share is hosted on an up-to-date TrueNAS server and accessed via SMB 3.1.1 and credentials stored in AD. The share is actually mounted, I can browse it from the server or from a device on the network and clips are correctly recorded to it, but the above keeps happening. Unfortunately, it results in two annoying/weird symptoms:

- The feed goes down for a while every time it happens (~20 seconds).
- A new mount is created but the old mount is not removed.

The second issue is possibly harmless but very strange indeed. After running the system for a while, there were over 5900 mounts to the same directory:

[image]

Strangely, each mount would randomly (?) use a different security protocol. Some use `ntlmv2`, some use `ntlmv2i`, and others use `ntlmsspi`. I believe this is related to an event in the `dmesg` log on the host:

```
[2404947.490327] CIFS: Attempting to mount //[redacted]/surveillance
[2404947.507674] CIFS VFS: Unable to select appropriate authentication method!
[2404947.507676] CIFS VFS: \\torres.home.hellstern.org Send error in SessSetup = -22
[2404947.510924] CIFS VFS: cifs_mount failed w/return code = -2
```

starbasessd, 2020-12-22:
Is your [redacted] a name or IP Address?

CIFS VFS: cifs_mount failed w/return code = -2

found this in Google:
https://www.google.com/search?client=firefox-b-1-d&q=CIFS+VFS%3A+cifs_mount+failed+w%2Freturn+code+%3D+-2

If you are using the name of the server, try switching to the IP address.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

dannytech, 2020-12-22:
It is a hostname, I'll try that out. Not sure why that would cause issues with unmounting, though, which seems to be a major part of the problem.

dannytech, 2020-12-23:
Unfortunately, that didn't solve it. I did notice that now it's trying to `umount` both the directory with the IP (which is what it's currently set to) and the old one with the hostname. There also appear to be more mount errors.

```
Dec 22 18:06:40 tuvok meyectl[10243]: mount error(16): Device or resource busy
Dec 22 18:06:40 tuvok meyectl[10243]: Refer to the mount.cifs(8) manual page (e.g. man mount.cifs) and kernel log messages (dmesg)
Dec 22 18:06:40 tuvok meyectl[10245]: mount error(2): No such file or directory
Dec 22 18:06:40 tuvok meyectl[10245]: Refer to the mount.cifs(8) manual page (e.g. man mount.cifs) and kernel log messages (dmesg)
Dec 22 18:06:40 tuvok meyectl[10247]: mount error(16): Device or resource busy
Dec 22 18:06:40 tuvok meyectl[10247]: Refer to the mount.cifs(8) manual page (e.g. man mount.cifs) and kernel log messages (dmesg)
Dec 22 18:06:40 tuvok meyectl[10249]: mount error(16): Device or resource busy
Dec 22 18:06:40 tuvok meyectl[10249]: Refer to the mount.cifs(8) manual page (e.g. man mount.cifs) and kernel log messages (dmesg)
Dec 22 18:06:40 tuvok meyectl[10253]: umount: /media/motioneye_[hostname]_surveillance_tuvok: no mount point specified.
Dec 22 18:06:40 tuvok meyectl[1531]:    ERROR: failed to unmount smb share "//[hostname]/surveillance" from "/media/motioneye_[hostname]_surveillance_tuvok"
Dec 22 18:06:40 tuvok meyectl[10254]: umount: /media/motioneye_[hostname]_surveillance_tuvok: no mount point specified.
Dec 22 18:06:40 tuvok meyectl[1531]:    ERROR: failed to unmount smb share "//[ip]/surveillance" from "/media/motioneye_[ip]_surveillance_tuvok"
Dec 22 18:06:42 tuvok motion[10303]: [0:motion] [NTC] [ALL] conf_load: Processing thread 0 - config file /etc/motioneye/motion.conf
Dec 22 18:06:42 tuvok motion[10303]: [0:motion] [NTC] [ALL] config_camera: Processing camera config file camera-1.conf
Dec 22 18:06:42 tuvok motion[10303]: [0:motion] [NTC] [ALL] motion_startup: Logging to syslog
Dec 22 18:06:42 tuvok motion[10303]: [0:motion] [NTC] [ALL] motion_startup: Motion 4.2.2 Started
Dec 22 18:06:42 tuvok motion[10303]: [0:motion] [NTC] [ALL] motion_startup: Using default log type (ALL)
Dec 22 18:06:42 tuvok motion[10303]: [0:motion] [NTC] [ALL] motion_startup: Using log type (ALL) log level (WRN)
Dec 22 18:07:06 tuvok motion[10303]: [1:ml1:Kes] [ERR] [ALL] motion_init: Error capturing first image
```

starbasessd, 2020-12-23:
I've had issues before with Proxmox, and unable to help user resolve. I'm
going to have to let someone else help you with it.
Current install instructions on other OSs, and the Docker image and manual
Docker installs work.
There is a user developing instructions for Kubernetes, but I can't get
Proxmox setup to work.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

dannytech, 2020-12-28:
I understand. I'm doing a bit of my own digging for the time being then. So far, I've noticed that the umount errors with "no mount point specified" are all trying to umount directories that don't exist (anymore)! Specifically, I'm using an AD user, so with a share of `\\[hostname]\surveillance` and an AD user called `tuvok@ad.[domain].org`, the correct folder format is `/media/motioneye_[hostname]_surveillance_tuvok_ad_[domain]_org`. However, `umount` is trying just `/media/motioneye_[hostname]_surveillance_tuvok`. When I was first setting this up, I tried the AD user without the `@` portion, which would result in the incorrect directory, but none of the current configuration files do this (I checked the backup files), so I'm not sure how it would remember that. It's also possible, I suppose, that motionEye is mounting to the correct folder but the `@` is breaking the umount and it's truncating to a nonexistent folder.

woody4165, 2021-02-10:
Hi

Did you solve?

I'm trying to use motioneye in an LXC container (Debian) in Proxmox, but when I reboot the CT the service is active, but the motioneye web page is not anymore reachable.
I've just opened am [issue](https://github.com/ccrisan/motioneye/issues/2028)

Thanks

starbasessd, 2021-02-10:
Have you tried any of these?
Google search: proxmox not releasing port used by container upon restart of
container
Lots of issues all over the place. most related to versions of proxmox...

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

woody4165, 2021-02-10:
Thank @starbasessd 

It was my fault.
I was restoring a backup with different paths inside camera configs and this was causing the issue.
But no message during the service start.

starbasessd, 2021-03-05:
If you are still having issues, please open a new issue.

closed 2021-03-05 by starbasessd