motioneye-project/motioneye #1976: Delayed stream of Reolink Ip camera model RLC-410W on motion eye

LimCC1234, 2020-12-24:
Hi sir, I uses the camera as mentioned above to display the stream using RTSP on motion eye but it have a high delay as the stream preview time didn't matches with the real world time. I had try lowering down the resolution and bitrate of the camera but it seems doesn't change much. Do you have any idea what causes this problem?

cap9qd, 2020-12-24:
I had lots of issues with my reolink cameras u til I swapped to the rtmp
stream. You may look to see if thst model supports rtmp and if it helps.

starbasessd, 2020-12-24:
Which Pi or other computer or SBC?
Are you using motionEye or motionEyeOS? 
Which version?
What BaseOS and version if motionEye?
Ethernet or Wifi for camera and Pi?
I use FosCam 9900 and RTSP and don't seem to have much issue with my production I5 server, Ubuntu 20.04, ethernet (and 4 other wifi network cameras. Running motionEye 0.42.1, and motion version 4.3.1
I can/do run in my testing environment any number of Pi1s, Pi2s, Pi3s & Pi3B+s, and several Pi4-4GB and Pi4-8GB and an untold number of PiZeros. I can reproduce most environments to help.

cap9qd, 2020-12-24:
https://forums.zoneminder.com/viewtopic.php?t=25874&start=15

LimCC1234, 2020-12-28:
> https://forums.zoneminder.com/viewtopic.php?t=25874&start=15
> […](#)

Thank you so much sir! RTMP also works fine for me, it seems smoother as compared o the RTSP.

LimCC1234, 2020-12-28:
> Which Pi or other computer or SBC?
> Are you using motionEye or motionEyeOS?
> Which version?
> What BaseOS and version if motionEye?
> Ethernet or Wifi for camera and Pi?
> I use FosCam 9900 and RTSP and [...]

Hi there, sir. I am currently using Pi4-8GB . motion eye installed inside Pi using the IOTstack, version not sure how to check (a newbie to linux system), both Pi and iP camera connected to the same ethernet. After changing to RTMP, the delay issue seems to be resolved.

closed 2020-12-28 by starbasessd