motioneye-project/motioneye #1977: Motioneye fails to start

Raymondgoff, 2020-12-24:
New install, gets to "Starting Motioneye: and eventually fails.

starbasessd, 2020-12-24:
motionEye or motionEyeOS? which version? (20190911, 20200203, 20200606, or
dev20201026 if motionEyeOS?
What device? Pi or other?
What network Connectivity?
What changes or files did you put into /boot prior to putting into device?
(if motionEyeOS?
It really does help to answer the questions.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Raymondgoff, 2020-12-24:
MotionEye OS - 2020606
Pi Zero W
Wireless - Changed the config in startup file to pickup AP, that works and says done 
thats the only changes

Raymondgoff, 2020-12-24:
Replies to ping while waiting for montioneye to start, and I see on network.  After couple min fails and restarts

starbasessd, 2020-12-24:
Try motionEye 20190911 or dev 20201026
Post your wpa_supplicant.conf file (redact the SSID and password)

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Raymondgoff, 2020-12-24:
country=US
        update_config=1
        ctrl_interface=/var/run/wpa_supplicant
        
        network={
            scan_ssid=1
            ssid="xx"
            psk="xx"
        }

Raymondgoff, 2020-12-24:
trying with dev 20201026........it worked

closed 2020-12-24 by starbasessd