motioneye-project/motioneye #1980: Docker Swarm Support

UncountedBrute3, 2020-12-27:
I really love your application, on Docker it has worked a treat so far.
I am scaling out my home Raspberry Pi 4 server to a farm and I was wondering whether MotionEye has support for Docker Swarm?

If it does, could you please guide me as to how to get it deployed in docker swarm?

starbasessd, 2020-12-27:
May I ask what your goal is?
What resources are you going to be using?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

UncountedBrute3, 2020-12-27:
I am using MotionEye to view other devices running MotionEyeOS. The MotionEye install will not be connected to a local camera setup, it is purely a hub to view other devices.

starbasessd, 2020-12-27:
Why use a swarm?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2020-12-27:
From what I'm reading, it takes 3 nodes for a swarm? 3 Pis, or 3 machines,
or 3 VMs?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

UncountedBrute3, 2020-12-27:
I am interested in using a swarm to learn docker swarm whilst applying it to an existing application I am using. If it's not recommended then that's not a problem.

starbasessd, 2020-12-27:
I'm just trying to figure out how to set up to be able to help you.
Swarms are primarily for load management, and I don't think motionEye
Dockerr fits the bill. I suppose you MIGHT need it in the event of multiple
(greater than 3) users watching the feeds at the same time, but tht's about
the only thing I can see 'swarming'... And again, you need multiple nodes
to connect to. So you'd need a couple of nodes, a bunch of systems to
connect to the swarm, etc...

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

UncountedBrute3, 2020-12-28:
Yes, I already have two pi's with docker swarm initialised. My use case is really around uptime. Say for example I wanted to bring down one of the nodes for maintenance, I could perform a DRAIN command on docker to remove the instance from the node, whilst leaving the website intact.
Because my Raspberry Pi cluster is a playground which can have nodes shutdown every so often for maintenance/swapping out, I thought having some redundancy would be helpful.

starbasessd, 2021-03-05:
If you are still having issues, please open a new issue.

closed 2021-03-05 by starbasessd