motioneye-project/motioneye #1981: Feature request: per-camera storage quota

marcgarciamarti, 2020-12-27:
It would be great to have the possibility to define storage quota complementary to already existing retention options (one day, one week, one month). This is useful specially when a false positive keeps triggering motion detection and footage starts piling up, not only on local storage but also in the cloud.

Some people are already using custom scripts to limit this. I wonder if you could evaluate adding an official feature to limit this.

```
Hi,

I have this crontab entry (sudo crontab -e):

*/20 * * * * /usr/local/bin/cleanold.sh

For the following script at /usr/local/bin/cleanold.sh:

#!/bin/bash
# Change ALERT to max usage percent, when exceeded the files will be deleted
ALERT=90
# Change FILESYSTEM to your motioneye data filesystem
FILESYSTEM=/
# Change MOTIONDATA to your motioneye data directory
MOTIONDATA=/var/lib/motioneye

USAGE=$(df -h --output=pcent "${FILESYSTEM}" | tail -1 | cut -d'%' -f1)
DAYFOUND=false
if [ ${USAGE} -ge ${ALERT} ]
then
        echo the used space is too high ${USAGE}, delete some files!
        for CAMERA in $(ls /var/lib/motioneye)
        do
                #echo ${CAMERA}
                DATA=$(ls "${MOTIONDATA}/${CAMERA}" | sort -r | tail -n 1)
                echo ${DATA}
                DATAS="${DATA}"$'\n'"${DATAS}"
                echo ${DATAS}
                DAYFOUND=true
        done
        if [ "${DAYFOUND}" = "true" ]
        then
                echo "${DATAS}" | sort | head -2 | tail -1
                DELETEDIR=$( echo "${DATAS}" | sort | head -2 | tail -1 )
                echo "${DELETEDIR}"
                for CAMERA in $(ls "${MOTIONDATA}")
                do
                        if [ -d "${MOTIONDATA}/${CAMERA}/${DELETEDIR}" ]
                        then
                                echo "${MOTIONDATA}/${CAMERA}/${DELETEDIR}"
                                #ls "${MOTIONDATA}/${CAMERA}/${DELETEDIR}"

# !!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
# WARNING IF YOU UNCOMMENT NEXT LINE DIRECTORIES AND FILES WILL BE DELETED
# !!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
                                rm -rf "${MOTIONDATA}/${CAMERA}/${DELETEDIR}"
                        fi
                done
        fi
fi
```

marcgarciamarti, 2022-02-26:
Hi, may I know if this feature request has been accepted?

thanks!

starbasessd, 2022-02-26:
Please note, we are looking for a new developer, as CCrisan has stepped down and put the project on hold until one is found. See #2290

JamBalaya56562, 2026-11-09:
Closing as a duplicate of #302, which asks for limiting the storage used by removing the oldest files. A per-camera limit fits there as part of the same feature, so please add it to #302.

closed 2026-11-09 by JamBalaya56562