motioneye-project/motioneye #1984: Extra Motion Options for Pi camera

123g-man, 2020-12-30:
I am running motionEye Version | 0.42.1, Motion Version | 4.2.2+gitUNKNOWN, OS Version | motionEyeOS 20200606 and all is working well. I am running with a MakerHawk Pi Cam (although I have also tried with a standard pi camera). I would like to change some of the camera parameters and see there is a box "Extra Motion Options" to enter some camera commands. Can anyone advise exactly what I need to enter here to say change the ISO, exposure, frame rate, etc? Nothing I've tried seems to work. Thanks

zagrim, 2020-12-31:
From the Motion docs for [`mmalcam_control_params`](https://motion-project.github.io/motion_config.html#mmalcam_control_params):

> In general, the command line options that are available via the raspistill / raspivid applications are the options that can be specified in this Motion configuration option.

I don't have Pi cam and I have never used one, but it seems that https://www.raspberrypi.org/documentation/raspbian/applications/camera.md (can be found via a link in the Motion docs) explains all the parameters. So it seems that the parameters you are looking for are `--ISO`, `--exposure` etc. I didn't find anything directly controlling the frame rate but I hope you can now figure that out.

123g-man, 2020-12-31:
Hello Zagrim,

Thank you for your reply. Yes, I've see that but I do not know the correct syntax to use in the motioneye "Extra Motion Options" box. For example, setting the ISO to 800 and the shutter speed to 1 second (to get good low-light image), I've tried;

ISO 800
shutter 1000000 
 
no effect, so

ISO = 800
shutter = 1000000

no effect, so

ISO,800
shutter,1000000

no effect, so

--ISO 800
--shutter 100000

no effect, so

... and so on, several other permutations.. Nothing I've tried seems to work and I can't find a description or example anywhere!

G

no change,

zagrim, 2020-12-31:
As I said, I have no possibility to try it out myself, but my guess for the correct format for the line to be put into "Extra Motion Options" (and which the comments in #575 seem to confirm) is:
`mmalcam_control_params --ISO 800 --shutter 100000`
or if Motion requires the `raspivid` options to be in the short format (with only one dash in the beginning) as all the examples I was able to find seem to use:
`mmalcam_control_params -ISO 800 -ss 100000`

The value might show wrapped to multiple lines in the text box in the MotionEye GUI, but as long as you don't put any linebreaks in it it should be fine.

closed 2021-03-05 by starbasessd