motioneye-project/motioneye #1985: Running As Normal User

jdenver13, 2021-01-01:
I install motioned and had everything working, I then followed the above procedure and now cant get the webpage running.

on entering the last line 
```
sudo -u youruser meyectl startserver -b -c /home/youruser/motioneye/motioneye.conf
```
I get "failed to configure login: [Errno 13] Permission denied: '/home/myname/motioneye/motioneye.log'"
When I cd to the /home/myname/motioneye/ directory, the only file in there is the modified motioneye.conf file.

Any though please.

starbasessd, 2021-01-01:
who owns and what are the permissions for /home/myname/motion?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

jdenver13, 2021-01-02:
Its easier if I copy and pasted the following
```
trevor@ubuntu:~$ ls -all
total 36
drwxr-xr-x 5 trevor trevor 4096 Jan  2 00:31 .
drwxr-xr-x 3 root   root   4096 Dec 20 09:48 ..
-rw------- 1 trevor trevor 1782 Jan  1 22:45 .bash_history
-rw-r--r-- 1 trevor trevor  220 Feb 25  2020 .bash_logout
-rw-r--r-- 1 trevor trevor 3771 Feb 25  2020 .bashrc
drwx------ 2 trevor trevor 4096 Dec 20 10:01 .cache
drwxrwxr-x 3 trevor trevor 4096 Jan  1 22:21 .local
-rw-r--r-- 1 trevor trevor  807 Feb 25  2020 .profile
-rw-r--r-- 1 trevor trevor    0 Dec 20 10:03 .sudo_as_admin_successful
drwxr-xr-x 2 root   root   4096 Jan  1 22:23 motioneye
trevor@ubuntu:~$ cd motioneye
trevor@ubuntu:~/motioneye$ ls -all
total 12
drwxr-xr-x 2 root   root   4096 Jan  1 22:23 .
drwxr-xr-x 5 trevor trevor 4096 Jan  2 00:31 ..
-rw-r--r-- 1 root   root   2905 Jan  1 22:23 motioneye.conf
trevor@ubuntu:~/motioneye$ 
```

jdenver13, 2021-01-04:
I would appreciate some help on this as I don't want to go changing permissions without knowing what I'm doing.

starbasessd, 2021-01-04:
Sorry, Real Life(tm) interfered with me building another machine to check
your issue. Will be doing so
today.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2021-01-04:
What OS did you build it on?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

jdenver13, 2021-01-04:
No worries, thank you. It's on Ubuntu 20.04 VM on Proxmox.
As I said, before it all worked until I tried to change to a 'normal user', I could access the UI on http://192.168.1.172:8765

starbasessd, 2021-01-04:
Just ran through instructions @
https://github.com/ccrisan/motioneye/wiki/Running-As-Normal-User
as my user:

mkdir /home/youruser/motioneye

cp /etc/motioneye/motioneye.conf /home/youruser/motioneye/motioneye.conf

nano /home/youruser/motioneye/motioneye.conf
     changed conf_path to /home/youruser/motioneye
     changed run_path to /home/youruser/motioneye
     changed log_path to /home/youruser/motioneye
     changed media_path to /home/youruser/motioneye
[ctrl]x y [enter]

sudo nano /etc/systemd/system/motioneye.service
     changed ExecStart=/usr/bin/meyectl startserver -c
/etc/motioneye/motioneye.conf
     to ExecStart=/usr/local/bin/meyectl startserver -c
/home/youruser/motioneye/motioneye.conf
added just below the above line
     User=youruser
[ctrl]x y [enter]

sudo systemctl daemon-reload
sudo systemctl restart motioneye

Of course change "youruser" to your user.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

jdenver13, 2021-01-04:
Thanks for the reply, I did follow those instructions and Ive gone back over them word for word and all's correct.
But still when I run the command:
```
sudo -u trevor meyectl startserver -b -c /home/trevor/motioneye/motioneye.conf
```
I still get the error:
```
failed to configure logging: [Errno 13] Permission denied: '/home/trevor/motioneye/motioneye.log'
```
As this is just a test case, in desperation I ran the command:
```
sudo chmod 777 -R /home
```
and the system in running now, not 100% sure this was the right thing to do, perhaps you can advise.

Thanks again.

starbasessd, 2021-01-04:
Just means everyone has full read/write/execute to every user folder, which
defeats the security that you were attempting to create. 7__ gives Owner
full control, _7_ gives the user's group full control and __7 gives
everyone else full control, which means any other user can write a file to
your folder, then run it from your folder, with no extra pris.
Definitely not good.
There are many ways to block access to that VM, and have only one user in
the sudoer's group, and only have log-on access via ssh & keys instead of
passwords...
Was there a particular reason to run as user instead of as system/root?

ls -la /home/myuser permissions for motioneye folder:
drwxrwxr-x  3 me me 4096 Jan  4 12:23 motioneye << effectively 775
ls -la motioneye
drwxr-xr-x  2  me me 4096 Jan  4 12:19 Camera1
-rw-r--r--  1 me me 2178 Jan  4 12:19 camera-1.conf
-rw-r--r--  1  me me  225 Jan  4 12:19 motion.conf
-rw-r--r--  1  me me 2901 Jan  4 11:57 motioneye.conf
-rw-r--r--  1  me me  589 Jan  4 14:40 motion.log
-rw-r--r--  1  me me    4 Jan  4 14:36 motion.pid
-rw-r--r--  1  me me  184 Jan  4 12:19 prefs.json
permissions for all the files is only 644

Did you do sysvinit or systemctl for startup?

I have discovered it doesn't start my camera and crashes when I use the
command line start, but works fine with the systemctl auto startup.
Weird
Checking further.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2021-01-04:
OK, it would seem manual mode and auto startup modes are mutually
exclusive.
If you put the auto startup stuff in, running the manual start will break
the auto start, and prevent the manual from running.
move the files from /etc/systemd/system/motioneye.service and
/etc/systemd/system/multi-user.target.wants/motioneye.service to your
/home/youruser folder,
reboot then run the manual start line.
I can't find where you can check it easily. You can run ps auxw|grep
meyectl and see it there.
You can also see if you can get to the WebGUI...

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

jdenver13, 2021-01-05:
> Was there a particular reason to run as user instead of as system/root? 

Not really, Ive now deleted the VM and re install Motioneye in a new VM without the 'Normal User' and it all works fine.

Thank you again for your time, help and advice on this issue.

Kind regards and stay safe.

closed 2021-01-05 by jdenver13

z0l0s, 2021-01-22:
Hi,
your motioneye directory is own by root but is inside your /home/trevor
drwxr-xr-x 2 **root   root**   4096 Jan  1 22:23 **motioneye**
and you want tu run motioneye demon with -u trevor, it's not compatible
i think you'l better chown this motioneye dir
chown trevor:trevor /home/trevor/motioneye