motioneye-project/motioneye #1988: Camera Names

mexjerry, 2021-01-04:
motionEye Version | 0.42
Motion Version | 4.2.2
OS Version | Ubuntu 18.10
Running Docker
I have set the Camera names in the webui, as Front, Back.
The .conf files contain Camera1, Camera2, rename cameras to match MY names, reboot device, they are renamed as Camera1, Camera2

ls -all motioneye/videos/
total 16
drwxr-xr-x 4 root  root  4096 Dec 30 12:33 .
drwxr-xr-x 4 jerry users 4096 Dec 30 12:30 ..
drwxr-xr-x 3 root  root  4096 Jan  4 07:04 Camera1
drwxr-xr-x 3 root  root  4096 Jan  4 07:05 Camera2

How to set this so it agrees with what I have set?
thanks

starbasessd, 2021-01-04:
edit camera-1.conf
line target_dir to whatever path/foldername you want
repeat as needed for the other camera .conf files
There isn't a way to modify in the WebGUI at this time.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

mexjerry, 2021-01-05:
That seems to have fixed it
Thanks
closed

closed 2021-01-05 by mexjerry