motioneye-project/motioneye #1989: Help with Motion Detection

apla2016, 2021-01-04:
Hello, thank you for great program, I am monitoring four cameras of different brands and was able to integrate video into my home automation software ioBroker.  But, I am struggling with motion detection setup. I configured one of my cameras to detect motion, see screenshot and expect to see images and video recorded when motion is detected by opening movie browser or picture browser, but they are always empty. Will appreciate any help. 

[image]

starbasessd, 2021-01-04:
10% of the image would have to change in each fram (LOTS of pixels) for it
to trigger. Try lower numbers (.3 to .1 percent is much more realistic.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

apla2016, 2021-01-04:
I tried 0.1 to 20% - same result, no motion detection. To simulate motion I am waving with my hand in front of the camera, affecting 30-50% of the image. What is actually the evidence that motion is detected? Logged images and videos in the system folder? Anything else? Thank you for your help.

starbasessd, 2021-01-04:
Turn on Show Frame Changes. It will draw a red line box around what it
thinks is moving. BTW, it's an inverse relationship between resolution and
percent motion, and try dropping minimum motion frames down to 2-3 until
you see results.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

apla2016, 2021-01-04:
Thank you, now I see that the motion is detected and also I realize that I have to enable recording images and pictures.

closed 2021-01-04 by apla2016

starbasessd, 2021-01-04:
Yep, that helps, too! It's a matter of playing with settings to fit your
environment, now...

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?