motioneye-project/motioneye #1990: update

titof2375, 2021-01-05:
Hello
I installed motioneye on an rpi 4 on the other hand it messed up funny and I would like to put motioneye on proxmox. i created an lxc and installed motioneye on it. on the other hand it is not up to date, on the line motion, it has version 4.1.1 instead of version 4.3.1 which is on the rpi. did you have a solution to update?

starbasessd, 2021-01-05:
What is your base OS for Proxmox, and which install instructions for
motionEye did you follow?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2021-01-05:
Are you using a pveam?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

titof2375, 2021-01-05:
> Quel est votre système d'exploitation de base pour Proxmox et quelles instructions d'installation pour motionEye avez-vous suivies?
> […](#)
> -- Thanks Kevin Shumaker Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel> N38° 19' 56.52" W85° 45' 8.56" Semper Gumby “Don't tell people how to do things. Tell them what to do and let them surprise you with their results.” - G.S. Patton, Gen. USA Ethics are what we do when no one else is looking. Quis custodiet ipsos custodes? “There is no end to the good you can do if you don’t care who gets the credit.” - C Powell You know we're sitting on four million pounds of fuel, one nuclear weapon and a thing that has 270,000 moving parts built by the lowest bidder. Makes you feel good, doesn't it?

I am going through the https://github.com/ccrisan/motioneye/wiki/Install-On-Debian

starbasessd, 2021-01-05:
Are you doing a full Debian, VM, or a pveam Debian, or ???  Which version
of Debian? There are a lot of variables here...

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

titof2375, 2021-01-05:
then lxc with debain 10

starbasessd, 2021-01-05:
I ran pveam available and got
debian-10-standard_10.7-1_amd64.tar.gz and a whole bunch of
debian-10-turnkeys
Standard or a turnkey version?
In order to help you, I need to know your setup. since there isn't a
process for the underlying system.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

titof2375, 2021-01-05:
I have the one the debian-10-standard_10.7-1_amd64.tar.gz

starbasessd, 2021-01-07:
Unfortunately I do not have any spare iron (read: extra PC) to toss ProxMox
onto) and I cannot get it set up under VirtualBox (my VM server) to allow
installation of the Ubuntu 10.7  CT.
I will therefor leave this in place an open to see if anyone else has
ideas, but there wasn't any other response to another user who had issues
with Proxmox and motionEye previously. Hopefully that will change.
I can support most Docker issues, and most VM issues, but not with Proxmox,
sorry.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

titof2375, 2021-01-10:
hello I just managed to put motion to the latest version on my lxc, because there is no privacy mask as on the version on the raspberry

titof2375, 2021-01-10:
on the rpi
[image]

on the lxc
[image]

starbasessd, 2021-01-10:
Currently running RaspberryPiOS 12/02/2020 on Rpi4-4GB with docker.io fully updated.
ran
sudo docker run --name="motioneye" -p 9000:8765 -p 9001:9001 -p 9002:9002 -p 9003:9003 -p 9004:9004 -p 9009:7999 --env="PUID=1234" --env="PGID=1234" --hostname="motioneye" -v /etc/localtime:/etc/localtime:ro -v /home/pi/local_drive/docker_motioneye/conf:/etc/motioneye:rw -v /home/pi/local_drive/motioneye_storage:/var/lib/motioneye:rw -v /home/pi/local_drive/docker_motioneye/log:/var/log:rw --restart="unless-stopped" --detach=true ccrisan/motioneye:dev-armhf;
to install motionEye in Docker for another user test and get:
[image]
Privacy mask in place with motionEye 0.42.1 and motion 4.2.2 (and functional)

titof2375, 2021-01-10:
I don't use docker myself

starbasessd, 2021-01-10:
I'll leave this here to see if anyone familiar with Proxmox can help you.

closed 2021-01-16 by titof2375