motioneye-project/motioneye #1991: Camera freezes and kernel issue

sdevrieze, 2021-01-06:
I have configured several camera's exactly the same, but one of them is causing troubles. The hardware in all camera's is almost exactly the same (only difference: other lens type but I would not suspect an optical item to cause freezes) and brand new.

After a few days the camera freezes (showing the same image and no updates). The logs include this line:
meyectl[310]:    ERROR: mjpg client timed out receiving data for camera 1 on port 8081

When I ssh into the camera and execute "sudo systemctl restart motioneye" it still does not work and the camera stream changes to grey with the text "unable to open video device". It starts working again only after reboot.

Last week I even could not log in into via ssh and had to manually unplug the device to reboot. At that time the web interface was still working and when trying to log in, I only saw the ssh fingerprint and not the command prompt. In the attached logs that I downloaded after the forced reboot, you can see there was some kernel issue.

Any idea what can be causing this and how to debug?

Hardware: Raspberry Pi 4 B+ 2GB with official HQ camera, an aluminium heatsink case from Pimoroni, a high endurance 256GB micro sd card, and a passively cooled PoE expansion board.

Software: latest Raspbian 10 with rpi-update to the latest version (kernel 5.10.3-v7l+), motionEye 0.42.1 and Motion 4.3.2.

sdevrieze, 2021-01-06:
[syslog.tar.gz](https://github.com/ccrisan/motioneye/files/5775469/syslog.tar.gz)

starbasessd, 2021-01-06:
Can you include the motioneye, and messages logs, please> You can get them
from Settings, Expert Settings.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

sdevrieze, 2021-01-06:
I guess this menu only exists in motioneEyeos? I am using motionEye and don't have that menu.

Anyhow, this currently are these logs (but no issues currently):

[0:motion] [NTC] [ALL] conf_load: Processing thread 0 - config file /etc/motioneye/motion.conf
[0:motion] [NTC] [ALL] config_camera: Processing camera config file camera-1.conf
[0:motion] [NTC] [ALL] motion_startup: Logging to syslog
[0:motion] [NTC] [ALL] motion_startup: Motion 4.3.2 Started
[0:motion] [NTC] [ALL] motion_startup: Using default log type (ALL)
[0:motion] [NTC] [ALL] motion_startup: Using log type (ALL) log level (WRN)
[1:ml1:Camera7] [WRN] [ENC] ffmpeg_set_codec_preferred: Preferred codec h264_omx has been blacklisted: Codec causes lock up on your FFMpeg version

[messages.gz](https://github.com/ccrisan/motioneye/files/5776108/messages.gz)

starbasessd, 2021-01-06:
What is your base os?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

sdevrieze, 2021-01-06:
Raspbian 10 as noted earlier.

starbasessd, 2021-01-06:
That doesn't help. There is a date attached to the image you burn, what is
that date?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

sdevrieze, 2021-01-06:
I have no idea as I used the Imager application.

starbasessd, 2021-01-06:
I would suggest re-imaging the Pi that shows the symptoms, or cloning a
known good SDCard from another Pi that is working, if available.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

sdevrieze, 2021-01-06:
Will that make a difference? As far as I can remember, I prepared all microsd cards the same evening.

starbasessd, 2021-01-06:
It may be as simple as that one has got some corruption or a bad memory
spot on the card. Either that or the Pi4 is defective, as the symptom also
would be caused by it overheating or the power supply not working
correctly. When the problem occurs, if you can ssh in, you can run the
command:
vcgencmd get_throttled [enter]
If it shows anything but 0x0, it's having issues.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

sdevrieze, 2021-01-08:
It has the problems again and it is showing 0x0. I doubt overheating would be the cause as I already reduced the framerate to 3 last time it happened. I see there are a new raspberrypi-kernel and rpi-eeprom version. I'll do the update and see if that helps...

starbasessd, 2021-01-08:
Then clone a working card, or re-image.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2021-03-05:
No response in >30 days. If you are still having issues, please open a new issue.

closed 2021-03-05 by starbasessd

sdevrieze, 2021-03-05:
This is still not fixed. I just updated packages, did rpi-update again and rebooted. So it is now again waiting until it breaks.

sdevrieze, 2021-03-06:
Just checked and it is broken again.

sdevrieze, 2021-04-02:
Ok, I completely replaced the camera with a new camera I was building at the start of this week. A few days later I checked and it had the same issue again. I switched to another port in the UniFi powered switch and up until now it is still running like a charm.

Is it technically possible that an unreliable power supply by the switch is causing this? If that is the case, maybe you should document this somewhere. If even an expensive PoE is not providing reliable power supply on some of its ports, there for sure will be other switches that will also cause these difficult to debug troubles.