motioneye-project/motioneye #1992: MotioneyeOS on Tinkerboard not finding local CSI camera

scott8, 2021-01-07:
I just tried this on both a Tinkerboard and Tinkerboard S.
I downloaded the image from https://github.com/ccrisan/motioneyeos/releases/download/20200606/motioneyeos-tinkerboard-20200606.img.xz and flashed it onto a micro sd card.
Plugged in the CSI camera to the tinkerboard with the silver connectors on the cable toward the HDMI port.
Booted up and open the web interface for motioneye.
No camera.
Adding a camera does not give me the Local camera type.
How do I troubleshoot this?
I also tried a new Raspberry Pi High Quality Camera with the same results.

scott8, 2021-01-07:
Correction. I get the Local V4L2 Camera option but no camera.
[image]

When I pull down the Camera selector I get 4 options:
[image]

I tried all 4. The best I can get is this:

[image]

This is with the new RPi High Quality Camera.

starbasessd, 2021-01-07:
On a Pi, you would have it auto-install as a mmal camera. I don't have a
tinkerboard to know if that is true or not to test.
Also, try dev20201026

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

scott8, 2021-01-09:
I decided to try the Tinker OS Debian distribution for this so I'd have the OS specific to this board, and install MotionEye and Motion on it.
I did all that and go the same behavior as above.
Then I found this page:
https://tinkerboarding.co.uk/wiki/index.php/CSI-camera-2.0.8
... where it says to run this to test the camera:

gst-launch-1.0 rkcamsrc io-mode=4 ! video/x-raw,format=NV12,width=640,height=480 ! rkximagesink

That works.

It also says to run:
media-ctl -p
and see the output for entity 7:

Media controller API version 0.1.0

Media device information
------------------------
driver          rkisp1
model           rkisp1
serial          
bus info        
hw revision     0x0
driver version  0.0.0

Device topology
<SNIPPED>
- entity 7: ov5647 2-0036 (1 pad, 1 link)
            type V4L2 subdev subtype Sensor flags 0
            device node name /dev/v4l-subdev2
pad0: Source
[fmt:SBGGR8_1X8/1296x972 field:none]
-> "rockchip-sy-mipi-dphy":0 [ENABLED]

So the camera does work on the Tinker Board S.

From here, can someone tell me the correct way to get Motion to work with it, and MotionEye?

Thanks,
Scott

starbasessd, 2021-01-09:
What options do you get when you select V4L2, rkisp1 (both listed)

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

scott8, 2021-01-09:
Unable to open Video Device shows up in the camera display, when I have the
motion daemon running as root at the command prompt. It is not running at
startup.

Should the bcm2835-v4l2 kernel module be present?

sudo lsmod
Module                  Size  Used by
8723bs               1757184  0

sudo insmod bcm2835-v4l2
insmod: ERROR: could not load module bcm2835-v4l2: No such file or directory

I'm upgrading the OS to see if that makes this work.

Scott

scott8, 2021-01-09:
Update, upgrade, and reboot did not help.
The module is still not available and I'm getting the same error. Unable to
Open Video Device.

scott8, 2021-01-09:
I found the URL for Motion and get this:
[image: MotionScreenShot1.jpg]

But this works, using /dev/video0, which is the device in my motion.conf
file.
The motion.conf file is stock, except I've turn off the web control local
only so I can hit the motion URL from my laptop.

gst-launch-1.0 rkcamsrc device=/dev/video0 io-mode=4 ! videoconvert !
video/x-raw,format=NV12,width=640,height=480 ! rkximagesink

scott8, 2021-01-09:
I found this thread which makes me think this is not going to work:
https://www.tinkerboarding.co.uk/forum/thread-492.html

I'm looking at some USB Dome cameras.

Scott

closed 2021-03-05 by starbasessd