motioneye-project/motioneye #1993: Video on motion detection has artifact

Tytouc, 2021-01-08:
Hi,
First time I'm posting here.
i'm using MotionEye on my raspberry pi 4 4GB under Buster (v10). (5.4.79-v7l+ #1373 SMP Mon Nov 23 13:27:40 GMT 2020 armv7l GNU/Linux)
I have 2 IP HD cameras.
I've setup motion detection for recording the video on one of the camera only.
When there is no motion, all is fine, but once there is a motion, the video contains 'artifacts'.
the typical 'artifact' is when someone in walking in the range of the video, it appears that the person is either 'frozen' and its 'shape' is not clear: it is blurry and with 'strange' colors.
On the video from which I extracted the pictures we can see 2 persons: one is frozen the other is 'blurry'.

(My Raspberry only run Ngix as a reverse proxy for SSL and MotionEye)

I've extracted a picture from the motion detection video. the scene is 2 persons leaving the blue car. the 1st one has been detected and got 'frozen'. The second one is 'blurry', and looks like 'transparent'.

On the picture, I have masked some private areas (black rectangles) and 'hidden' part of the car.
You can see motion detection area in the red rectangle from MotionEye around the person being 'frozen', and I've added a green rectangle for the second person being 'blurry/tranparent'.

Any dear I could investigate and solve the issue?

[image]

starbasessd, 2021-01-08:
Which install instructions for motion/motionEye?
What version of motion?
What version of motionEye?

Tytouc, 2021-01-08:
Hi,
I've installed motioneye following this site: https://github.com/ccrisan/motioneye/wiki/Installation
Installed Motion version is 4.2.2
Motioneye version 0.42.1

Thanks,

starbasessd, 2021-01-08:
Can you try updating/upgrading motion to 4.3.1?

wget https://github.com/Motion-Project/motion/releases/download/release-4.2.2/pi_buster_motion_4.2.2-1_armhf.deb
 dpkg -i pi_buster_motion_4.2.2-1_armhf.deb

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Tytouc, 2021-01-08:
sure, but isn't you wget / dpkg command referring to 4.2.2 instead of 4.3.1?

starbasessd, 2021-01-08:
Yeah, yeah, yeah... Details <grin> Clicked send before I was ready, sorry.
Lack of caffeine this morning...

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Tytouc, 2021-01-08:
haha, it's evening already here ;)
Do you have the correct commands for 4.3.1?
shall I restart either the Pi or MotionEye or anything else after the install?

starbasessd, 2021-01-08:
wget https://github.com/Motion-Project/motion/releases/download/release-4.3.1/pi_buster_motion_4.3.1-1_armhf.deb
<https://github.com/Motion-Project/motion/releases/download/release-4.2.2/pi_buster_motion_4.2.2-1_armhf.deb>
dpkg -i pi_buster_motion_4.3.1-1_armhf.deb

 pip install motioneye --upgrade
 systemctl restart motioneye

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Tytouc, 2021-01-08:
Thanks,
I ran the commands (wget, dpkg, pip and systemctl): motion -version shows 4.3.1 now.
FYI: ffmpeg version is 4.1.6-1~deb10u1+rpt1

I'll see if it improves only tomorrow, since it is night now, and I'd prefer doing tests is similar conditions.
I keep you posted.

Cheers,

cap9qd, 2021-01-08:
I have had something similar to this issue before with my ReoLink cameras
before swapping the stream to the rtmp stream. Apparently reolink's rtsp
stream implementation was lacking and since their NVR uses the rtmp stream
they never fixed it. Might be worth seeing if your cameras have a different
stream you can try.

Tytouc, 2021-01-10:
Hi,
so, test result: is..; a bit better, but, still some 'blurry' pixels on the moving 'object' and frame rate is very very slow as soon a a motion is detected.
I'm using the RSTP stream. The camera proposes the RTMP. I can check with this protocol if I'm able to configure it. So far, I've tried RTMP but wihtout removing the existing camera in MotionEye and I cannot get the live view. Maybe the cam cannot have both RTSP and RTMP connected at the same time.
In the case I can configure it I'll come back to you with news.
FYI: My IP cam is a iEgeek HD.

Tytouc, 2021-01-10:
I'm not able to configure MotionEye to use the RTMP protocol. IE80 states a specific port for RTMP, but I'm not able to find the proper URL, hence I used the same as for RTSP, except I used the RTMP port. I do not see any image :( I've 1st removed the camera from MotionEye.
I'll thus come back to the RTSP protocol, but I'll try with the second stream of the camera which is of lower quality to see it it improves.

starbasessd, 2021-01-10:
Which camera9s)? Did you look at:
https://www.ispyconnect.com/sources.aspx
to see if they have suggestions?

Tytouc, 2021-01-10:
Hi, this is the ieGeek IG80, I went to this site, but there is only the RTSP URL, not the RTMP.
Appart from videos, the still images are also showing the issue I'm facing. here it is (hiding some perso stuff behind the black areas).
The scene is 2 persons arriving home. You can see that the 2 persons are at 2 different places on the same picture. and the 'shape' of the persons are not that clear.
This is a still image captured during motion.
This was captured once I have updated Motion to 74.3.1 as asked by @starbasessd and nothing else modified.

[image]

zagrim, 2021-01-11:
Still images saved due to detected motion are taken from the same set of frames the motion videos are compiled by Motion, so based on that it makes sense to see the same artefacts in both.

@Tytouc you said "still some 'blurry' pixels on the moving 'object' and frame rate is very very slow as soon a a motion is detected", which makes me wonder if it is at least partly due to the video de/encoding being too heavy for the Pi (frames being dropped can cause weird artefacts, with the actual appearance and persistence of them depending on the codec being used). Please check the CPU load of the Pi during motion event. If the CPU load goes through, you might try to use a significantly lower resolution and/or framerate for the camera under "Video Device" settings and see if that makes the issue go away.

Tytouc, 2021-01-11:
My config in Video Device is 800x600 20fps.
without motion detected TOP gives: (with 'I' option). I'll try to have the values during a motion detection.

top - 09:52:56 up 11 days, 15:03,  1 user,  load average: 2,36, 2,33, 2,18
Tasks: 148 total,   1 running, 147 sleeping,   0 stopped,   0 zombie
%Cpu0  :  1,0 us,  0,0 sy,  0,0 ni, 99,0 id,  0,0 wa,  0,0 hi,  0,0 si,  0,0 st
%Cpu1  : 30,0 us,  8,4 sy,  0,0 ni, 61,6 id,  0,0 wa,  0,0 hi,  0,0 si,  0,0 st
%Cpu2  : 91,4 us,  8,6 sy,  0,0 ni,  0,0 id,  0,0 wa,  0,0 hi,  0,0 si,  0,0 st
%Cpu3  :  0,3 us,  0,3 sy,  0,0 ni, 99,3 id,  0,0 wa,  0,0 hi,  0,0 si,  0,0 st
MiB Mem :   3775,5 total,    588,0 free,    391,8 used,   2795,8 buff/cache
MiB Swap:    100,0 total,     79,2 free,     20,8 used.   3101,2 avail Mem

  PID USER      PR  NI    VIRT    RES    SHR S  %CPU  %MEM     TIME+ COMMAND
 6070 root      20   0  297424  93368  26228 S  33,1   2,4   9:13.09 motion
21577 root      20   0   71368  26696  10216 S   0,7   0,7 145:40.14 meyectl
 ...

Tytouc, 2021-01-11:
I got htop during motion detection. Motion went at max @220% (total on 4 cores) (~240% for all processes)
[image]

Tytouc, 2021-01-11:
on the test above, the video was ok regarding the 'strange 'pixels' even if it was like missing some frames. The scene was with 1 person only vs 2 persons in the previous motions. I'll test with 2 persons to see if detecting 2 motions/2 motion area may change the behaviour. I'll run the test with a lower FPS: 5 instead of 20.

zagrim, 2021-01-11:
Is the last screenshot (at 12:06:30) with or without motion? 
There the 1 minute average of total CPU load (the first figure after "load average" on the 1st row) is almost 4 (which is the maximum with 4 cores) and also the 5 minute average is still near to 4. That tells that the CPU capacity likely has been briefly maxed out more than once during the previous five minutes, even though the current load of the CPU cores on the instant of taking the screenshot isn't anymore that bad. Anyway, the overall load is the key, not the load caused by `motion` alone.

The other screenshots look not so bad looking at the load averages, although it might not be possible to pinpoint the limit of video processing performance by just looking at the CPU load. The one from 11:39:51 shows 34.4% system load in addition to 53.6% user load totalling 88% (or only about 0.5 cores worth of idle CPU left). But the load needs not hit the roof for things starting to lag (and frames beginning to drop), since the kernel prioritises system processes over the user processes such as `motion`, and even a process as light as `top` stops responding and updating the info in the terminal before the overall CPU load has maxed out. Video processing starts to drop frames heavily long before that due to the CPU load being rather non-stable (with bursts of load interleaving and overlapping each other).

Anyway, based on this evidence I think the CPU load might still not the be _only_ reason for the artefacts, since when there's roughly 1 CPU core being idle I don't think frame dropping should be occurring. At least in theory :smile:

Tytouc, 2021-01-11:
Hi screenshot at 12:06:30 is without motion, bit there were motion maybe in the past 5 minutes.
I need to run the tezy with 2 persons and a framerate at 2fps.
Is ther any motion or motioneye log I could activate (and how?) that could help?

Tytouc, 2021-01-12:
test with several persons in motion and with a framerate of 5fps are better, but still I have these strange pictures and some frames are dropped.
No idea on how to solve it :(

zagrim, 2021-01-13:
Unfortunately I'm also out of ideas now... Are there any options on the camera configuration (and I mean on the camera, using some sort of management/configuration interface) where you could try to change the encoding/codec it is using for the video stream? That might be worth to try in case Motion has issues decoding the output of that particular implementation of the codec.

Tytouc, 2021-01-13:
Hi, yes, there are different settings I could configure on the camera interface. I have kept the default ones as here under.
[image]

I have the following choices:
- video format: 50Hz or 60Hz
- Video Coding : H264 base profile, Main profile or High profile, or H265 main profille
- Max frame rate: 1 to 30
- bit rate: CBR or VBR
- Image quality: 1 to 6

No idea what to test here, I do not understand all the parameters and I didn't find the explanation of the same in the camera documentation :(
I can try the H265 profile since it may use a different codec vs the H264 based ones.

starbasessd, 2021-01-13:
I recommend
Follow your video type: PAL = 50, NTSC = 60
Max Framerate < 15
Max kbps < 1500
CBR (Constant Bit Rate)
Image Quality 1, 2, or 3

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

zagrim, 2021-01-15:
H265 is newer than H264 and does more aggressive compression (with better image quality, too), and is also much more CPU intensive to en/decode when there's no hardware support for it available (such as in the RPi). 

I don't know if it makes any significant difference but you might try H264 Main profile (instead of High), I'd assume it to produce somewhat larger files and possibly lose some of the image quality, but as far as I've understood the profiles right, Main profile should require less computation to en/decode than High, and probably the algorithm implementation being used is also somewhat different between those so switching that setting might have some effect on the artefacts.

The rest of the settings influence mostly the quality vs size trade-off and will effect both the CPU load as well and image quality and video file size, but they should affect the artefact issue only via the CPU load.

Tytouc, 2021-01-15:
Thanks @zagrim for your answer. I'll test H264 Main Profile. I'll come back to you once I have a significant test.
Rgds,

Tytouc, 2021-01-18:
Hi, I've tried H264 Main profile. Result is the same as soon as there are 2 persons on the motion. Maybe 2 camera is too much for the PI4 even when the motion is detected only on 1 camera ... I wanted to add 1 or 2, but it is maybe not reasonable.
The option could be to have 1 Pi for each camera for the motion detection, and 1 Pi for supervision only. But, it won't be the same budget, and a more complex setup.
thanks

starbasessd, 2021-01-18:
I have 5 cameras being used, 3 D-Link DCS-5020L, 1 FosCam 9XXX rtsp
(network), and a direct Connect USB TIGA generic DashCam, all being motion
detected, on Pi3B+ and Pi4-4GB on a regular basis.for testing. I can tossd
in an occasional PiZero with PiCam (regular or Fast Network Cam setup). I
Use modes without OMX due to issues upstream with motion.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Tytouc, 2021-01-18:
@starbasessd thanks for your inputs.
I'm not sure I understand properly.
Ar you saying that your 3x D-Link + Foscam + USDB cam are connected all to the same Pi3 or Pi4?
In such case, either I have too high expectations in term of image quality (either from Motion settings or directly from the camera) or I have bad configurations.
My Pi4 4GB is only hosting MotionEye + Ngix as a proxy (to do SSL endpoint in order to ensure user authentication and encprytion when I access it from outside my LAN: secure the login/pwd as an example).
I'm not an expert at all in Linux/Video/motion etc, but I have quite good basis in security etc ;)

I'll continue to try lower profiles/configuration to see if I can reach something acceptable.
Thanks again for your support.

starbasessd, 2021-01-18:
Yes, they are all connected to the same PI3 or Pi4 for re-creating user
issues.MotionEyeOS works the best ( completely stripped out OS by CCrisan)
Works good enough on RaspberryPiOS Lite minimal install by itself with
motion/motionEye on Pi3 & Pi4.
When I start adding things to it, like desktop, webservers and other
services, it rapidly bogs down. Toss in encryption services, and boom...
To do those things, I have a Lenovo M73-i5 TFF, running Ubuntu Server 20.10
minimal install, and even there, it isn't happy. (I have another TFF for my
production motionEye hub.)
I woulod drop your video to 800x600 and less than 10fps if you are tyring
to do it all on the Pis

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Tytouc, 2021-01-18:
Hi @starbasessd thanks for your inputs.
I have a 'desktop Raspberry Pi OS' base even if it is currently running headless, but I prefer to be able to access it through VNC, hence the desktop.
My settings in MotionEye are already 800x600 and frame rate set @5.
I cannot remove 'encryption' otherwise: no way I enter credentials through a non-secure HTTP interface.

So, I'll try other setting, lowering the quality etc... and in case I cannot succeed I may start thinking in adding 1x RPI4 for CCTV/Motion only (based on MotionEyeOS) and a second one for this Ngix/SSL stuff, and maybe hosting other projects I may have. But, really, I'd prefer to have only 1 pi to manage ;)

I may also have a look on another solution based on Shinobi, but it seems very less flexible, especially because I'm sending media to a NAS over FTP (and set "keep files" for a 'short' duration to do not fill my SSD too fast). I'm also using email notifications. Not sure I can do all that with Shinobi.

That's a pity I cannot make it work. I based my 'build' on several web searches and I didn't notice such problem will occur :(

Tytouc, 2021-01-18:
BTW: what mean the "nb1/nb2 fps" on the video in MotionEye when I click on th picture?

starbasessd, 2021-01-18:
streaming and recording frames per second.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Tytouc, 2021-02-04:
Thanks @starbasessd.
Sorry to have been mute since many days, but I was too busy with my job ;)
I changed some settings of MotionEye to access the second stream of the cameras. The cameras have a second stream which video size is hard-coded to 640x352. This has reduced the CPU usage of motion process to around 160% (vs 250% in the previous config). Note that CPU grows according to the number of cameras. So, no way I can had a third camera as I plan to do. I've also configured the PI to boot on console and not on desktop to minimize resource allocation. (I've not removed the desktop components, but removed VNC). I've also configured the 'Movie' section as 'Movie passthrough', disabled video streaming and text overlay.
The movie of a detected motion seems to do not have anymore any artifact, but still, this movie is showing a kind 'picture-by-picture' when viewed, jus as it was a slow-motion.
Maybe i'm missing some hardware acceleration ? I have no idea on how to check this. any idea?.

Thanks,

starbasessd, 2021-02-04:
The percentage, if divided by 4, would show the average CPU usage
(otherwise you are getting all 4 cores added together).
As to the acceleration, that is more of a @jasaw question...

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Tytouc, 2021-02-04:
Hi, yes, 160% is 40% of CPU (for 4 cores). I didn't that the %CPU went close to 400% (100% avg) during motion detection, so, not sure this is the issue that produces a 'slow-motion' movie. Moreover, I understood that the 'passthrough' parameter should not change the input stream encoding, so it should be the same movie as the camera is streaming. But this is not the case. I've added 'motion detection' from the camera itself, and have compared the movie generated/stored by the camera and the one from Motion. The one from the camera is 'smooth'.
At the end, I'm not happy with the quality: 640x352 is definitely not enough to see details.
Maybe an option could be to use the camera motion detection and not motion. The only issue here comes from my camera: it produces a raw H.264 video which requires to be encapsulated into a container (MKV or MP4..) to be readable by 'standard' software. I did such conversion, but it misses the sound. Using the P2P mobile app of the camera I can download the movie in MP4 with sound. I do not want P2P since it goes though a 3rd party server which I do not know, and for which I have no reason to trust their security.
If I use Camera Motion, then, it can send email notification. Maybe, I could send an email to 'motion/Motioneye/FFMPEG or anything else' so that it gets the captured video from the camera (HD stream) and runs a 'conversion' to make it MKV/MP4 with sound?
In such case I'd use MotionEye only for live viewing (small size stream) of all cameras on a single place

starbasessd, 2021-02-04:
Recording with sound is one of the most asked for 'enhancements' already.
There are several people in the Issues that have written/modified to do so,
might want to check them out.
motionEye/motionEyeOS is written to be compatible with many different
systems, and not truly  optimized for all the hardware types out there.
Ditto for Motion-Project & ffmpeg...

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2021-03-05:
No further response in 30 days. If you are still having issues, please open a new issue.

closed 2021-03-05 by starbasessd