motioneye-project/motioneye #1994: Getting warning for platform not match when running ccrisan/motioneye:dev-armhf docker image with Raspberry Pi 2

mcgyver83, 2021-01-09:
Running ccrisan/motioneye:dev-armhf docker image on raspberry pi2 B I have this warning despite of I'm asking for dev-armhf image.
This started a couple of weeks ago.

docker run --name="motioneye" \
  	        -p 9000:8765 \
		-p 9001:9001 \
		-p 9002:9002 \
		-p 9003:9003 \
		-p 9004:9004 \
		-p 9009:7999 \
		--env="PUID=1234" \
    	        --env="PGID=1234" \
		--hostname="motioneye" \
		-v /etc/localtime:/etc/localtime:ro \
		-v /home/pi/local_drive/docker_motioneye/conf:/etc/motioneye:rw \
		-v /home/pi/local_drive/motioneye_storage:/var/lib/motioneye:rw \
		-v /home/pi/local_drive/docker_motioneye/log:/var/log:rw \
		--restart="unless-stopped" \
		--detach=true \
		ccrisan/motioneye:dev-armhf;

WARNING: The requested image's platform (linux/amd64) does not match the detected host platform (linux/arm/v7) and no specific platform was requested

OS: Raspbian GNU/Linux 10 (buster)
Client: Docker Engine - Community
 Version:           20.10.2
 API version:       1.41
 Go version:        go1.13.15
 Git commit:        2291f61
 Built:             Mon Dec 28 16:19:01 2020
 OS/Arch:           linux/arm
 Context:           default
 Experimental:      true

Server: Docker Engine - Community
 Engine:
  Version:          20.10.2
  API version:      1.41 (minimum version 1.12)
  Go version:       go1.13.15
  Git commit:       8891c58
  Built:            Mon Dec 28 16:16:42 2020
  OS/Arch:          linux/arm
  Experimental:     false
 containerd:
  Version:          1.4.3
  GitCommit:        269548fa27e0089a8b8278fc4fc781d7f65a939b
 runc:
  Version:          1.0.0-rc92
  GitCommit:        ff819c7e9184c13b7c2607fe6c30ae19403a7aff
 docker-init:
  Version:          0.19.0
  GitCommit:        de40ad0

starbasessd, 2021-01-09:
Do you have the "\" at the end of each line?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

mcgyver83, 2021-01-09:
Yes, to have better indentation in the script that contains this docker run command.
Is the standard way to have uno single command line split. Isn't it?

starbasessd, 2021-01-09:
Yes, the '\' tells the system that it's all one line.
Checking. Not able to use a Pi 2B, currently both in use. Will test with
Pi4. What is your BaseOS? (Which version did you download? (Date stamp in
filename)

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2021-01-10:
RaspberryPiOS image:20201202, full updated to tonight.
Installed docker:
apt install docker.io
rebooted
Ran:
pi@raspberrypi:~ $ sudo docker run --name="motioneye" -p 9000:8765 -p
9001:9001 -p 9002:9002 -p 9003:9003 -p 9004:9004 -p 9009:7999
--env="PUID=1234" --env="PGID=1234" --hostname="motioneye" -v
/etc/localtime:/etc/localtime:ro -v
/home/pi/local_drive/docker_motioneye/conf:/etc/motioneye:rw -v
/home/pi/local_drive/motioneye_storage:/var/lib/motioneye:rw -v
/home/pi/local_drive/docker_motioneye/log:/var/log:rw
--restart="unless-stopped" --detach=true ccrisan/motioneye:dev-armhf;
Unable to find image 'ccrisan/motioneye:dev-armhf' locally
dev-armhf: Pulling from ccrisan/motioneye
e34ede4dd6ed: Pull complete
5425b2beb7b9: Pull complete
da84edc035ad: Pull complete
480513a66d75: Pull complete
679906dfb021: Pull complete
Digest:
sha256:2a1978bcaee4102f094ce8f7ee9b67261b86460cff00a1250d74345b26e9df02
Status: Downloaded newer image for ccrisan/motioneye:dev-armhf
e964b738099d4c704b972c16a78aa976dc15648b25e4bdf7da93ca60e18395a2
pi@raspberrypi:~ $
Opened browser on same network, http://[ip_address_of_pi]:9000
I get the standard logon for motioneye.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

zagrim, 2021-01-10:
Isn't this the same thing as in #1897 even though here it's about dev images, not master? (I don't have any solution, just remembered there was similar conflict reported earlier)

mcgyver83, 2021-01-10:
> […](#)
> -- Thanks Kevin Shumaker Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel> N38° 19' 56.52" W85° 45' 8.56" Semper Gumby “Don't tell people how to do things. Tell them [...]

OS: Raspbian GNU/Linux 10 (buster)
Docker client 20.10.2
Docker server 20.10.2

mcgyver83, 2021-01-10:
> Isn't this the same thing as in #1897 even though here it's about dev images, not master? (I don't have any solution, just remembered there was similar conflict reported earlier)
> [image]

Yes looks like #1897 .
I'm starting having this issue also with portainer docker image running fine before.

starbasessd, 2021-01-10:
A minor exception: The command line @mcgyver83 posted is the "ccrisan/motioneye:dev-armhf;" version, as is the command line I posted.
I removed that image, and installed the master-armhf in the same environment (RaaswpberryPiOS 12/02/2020 & docker.io both fully updated in the RPi4-4GB)
ran the same command as above, but with master-armhf instead of dev-hf and got the same working results.
pi@raspberrypi:~ $ sudo docker run --name="motioneye" -p 9000:8765 -p 9001:9001 -p 9002:9002 -p 9003:9003 -p 9004:9004 -p 9009:7999 --env="PUID=1234" --env="PGID=1234" --hostname="motioneye" -v /etc/localtime:/etc/localtime:ro -v /home/pi/local_drive/docker_motioneye/conf:/etc/motioneye:rw -v /home/pi/local_drive/motioneye_storage:/var/lib/motioneye:rw -v /home/pi/local_drive/docker_motioneye/log:/var/log:rw --restart="unless-stopped" --detach=true ccrisan/motioneye:master-armhf;
Unable to find image 'ccrisan/motioneye:master-armhf' locally
master-armhf: Pulling from ccrisan/motioneye
eaaa43e22ddc: Pull complete
38b5f9df17bb: Pull complete
5a36f2cd0f41: Pull complete
957aa23b13ac: Pull complete
1b23938bf6aa: Pull complete
Digest: sha256:1d7abea3e92e8cc3de1072a0c102a9d91ef2aea2a285d7863b1a5dadd61a81e1
Status: Downloaded newer image for ccrisan/motioneye:master-armhf
57277e82e6c1398cb197a269e7954488f2c66dab2bb565bf7eaeee72827adc82
pi@raspberrypi:~ $
and is working

as an aside, the dev-armhf image is 11 days old, the master-armhf is 7 months old as of 01/10/2021...
pi@raspberrypi:~ $ sudo docker version
Client:
 Version:           18.09.1
 API version:       1.39
 Go version:        go1.11.6
 Git commit:        4c52b90
 Built:             Fri, 13 Sep 2019 10:45:43 +0100
 OS/Arch:           linux/arm
 Experimental:      false

Server:
 Engine:
  Version:          18.09.1
  API version:      1.39 (minimum version 1.12)
  Go version:       go1.11.6
  Git commit:       4c52b90
  Built:            Fri Sep 13 09:45:43 2019
  OS/Arch:          linux/arm
  Experimental:     false

Which version of docker did you install?

mcgyver83, 2021-01-10:
> A minor exception: The command line @mcgyver83 posted is the "ccrisan/motioneye:dev-armhf;" version, as is the command line I posted.
> I removed that image, and installed the master-armhf in the same [...]

I wrote in the first comment that I'm using Docker 20.10.2 for client and server.
You are testing with a Rpi4 that is ARM64 but I'm reporting that issue using a rpi2 that is armhf.
Looks like that docker 20.10.* now does extra checking for the architecture meta of containers.

starbasessd, 2021-01-10:
By which version, I meant where did you install it from. I installed docker
from command line:
sudo apt install docker.io

Where did you get your version of docker, what commands did you use to
install docker?

On Sun, Jan 10, 2021 at 8:58 AM mcgyver83 <notifications@github.com> wrote:

>
> I wrote in the first comment that I'm using Docker 20.10.2 for client and
> server.
> You are testing with a Rpi4 that is ARM64 but I'm reporting that issue
> using a rpi2 that is armhf.
> Looks like that docker 20.10.* now does extra checking for the
> architecture meta of containers.
>
> —
>

mcgyver83, 2021-01-10:
> By which version, I meant where did you install it from. I installed docker from command line: sudo apt install docker.io Where did you get your version of docker, what commands did you use to install docker?
> […](#)
> On Sun, Jan 10, 2021 at 8:58 AM mcgyver83 ***@***.***> wrote: I wrote in the first comment that I'm using Docker 20.10.2 for client and server. You are testing with a Rpi4 that is ARM64 but I'm reporting that issue using a rpi2 that is armhf. Looks like that docker 20.10.* now does extra checking for the architecture meta of containers. —

Sorry, now I got you question. Right now I'm using docker provided with OpenMediaVault 5 for raspberry but I have the same WARN with docker installed via `curl -fsSL https://get.docker.com -o get-docker.sh
`

starbasessd, 2021-01-10:
OK, will test that in a few...

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

mcgyver83, 2021-01-10:
> OK, will test that in a few...
> […](#)
> -- Thanks Kevin Shumaker Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel> N38° 19' 56.52" W85° 45' 8.56" Semper Gumby “Don't tell people how to do things. Tell them what to do and let them surprise you with their results.” - G.S. Patton, Gen. USA Ethics are what we do when no one else is looking. Quis custodiet ipsos custodes? “There is no end to the good you can do if you don’t care who gets the credit.” - C Powell You know we're sitting on four million pounds of fuel, one nuclear weapon and a thing that has 270,000 moving parts built by the lowest bidder. Makes you feel good, doesn't it?

Many thanks. As said I'm not sure is something related directly to motioneye due the fact that I have this issue with official portainer docker image but not with linuxserver/plex image.
Maybe something related to image build.

starbasessd, 2021-01-10:
OK, I ran the docker install
I ran the armhf image pull.
I got the warning.
I basically ignored the warning
I ran (after adding pi to docker group and re-logging in)
docker start motioneye
Works.
I did a google on the warning.
Happens for portainer as you noticed. I do not know if it prevents
portainer from working.
motionEye still works as expected. Just have to run it as above.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2021-01-10:
https://www.google.com/search?client=firefox-b-1-d&q=WARNING%3A+The+requested+image%27s+platform+%28linux%2Famd64%29+does+not+match+the+detected+host+platform+%28linux%2Farm%2Fv7%29+and+no+specific+platform+was+requested

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

mcgyver83, 2021-01-10:
Yes, sorry, I forgot to add that motioneye (and portainer) runs fine despite of this WARN.

Is a bit annoying the WARN but no functional issues

starbasessd, 2021-03-05:
No further response in >30 days. 
It looks like it is just how it's posted/labeled in Docker (arm64/armhf/amdxx/etc) 
If you are still having issues, please open a new issue.

closed 2021-03-05 by starbasessd