motioneye-project/motioneye #1996: Picture Browser Crashing?

carlbeech, 2021-01-11:
Hi

Many thanks for a great utility!

I've just performed a fresh installation of motioneye onto a ubuntu PC - with a RTSP camera, using 1280x720 pictures.

The camera is pointing at the front garden, and unfortunately it has a number of plants - which means using motion detection isn't really a good option (e.g. using a mask, I'd lose significant portions of the picture). I've got a good size hard drive, and enough space to hold 1 frame per second, with a retention of a week.

While I'm able to store the pictures, when I click on the picture browser, I get the wait animation for a few seconds and then it just goes away - the browser never appears. I've no doubt that its because I've got too many pictures (3600 per hour)... 

Is there any way to get around this issue?  I thought I'd change the directory format from %Y-%m-%d/%H-%M-%S to %Y-%m-%d-%H/%H-%M-%S - thinking that it'll just show the last directory contents - or would 3600 be too many?

Please let me know if there's any logs that you'd need to help diagnose (and where to locate them)

I'm aware I could use a 3rd party directory picture gallery app, but I'd prefer to keep all the functionality within motioneye...

Many thanks

Carl.

starbasessd, 2021-01-11:
SSH or console into the motionEey PC, and run command:
sudo nano /etc/motioneye/motioneye.conf [enter]
find the line that begins:
list_media_timeout
change the number upwards. I have mine set at 120 right now, and seems to
work for me.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

carlbeech, 2021-01-11:
Hi

Thanks for the advice - seems the default is 120? - I tried increasing this to a ridiculous value, but it didn't have any impact - after a while the whirling wait symbol just went away and there was no browser coming up...

Any other thoughts..?

Cheers
Carl.

starbasessd, 2021-01-11:
if you delete or move all the pics out of that folder tree, and test every
few minutes, does it ever work for you?
Does it stop working after 10 minutes? 30 minutes? An hour?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

carlbeech, 2021-01-11:
It does work (it was working when I first created the install - however, I'm not sure what the 'breakpoint' number would be...
I can give that a go and let you know.
Cheers

Carl

starbasessd, 2021-01-11:
Once you find that number, you can, say, every hour, move the files that
are not current, to an archive folder tree. The reason that number, and the
email listing number are there is because it looks at ALL the files in that
tree, every time, and it's no where near instant. When it has too many
files to look through, it gets grumpy and gives up. Not one of the best
features...

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

carlbeech, 2021-01-11:
Yep - could do that - saying that, now I've moved everything into a different folder, its working again (as expected, as there's no files there) - I've been moving the 'hourly' folders into the camera1 folder (I'd changed the folder format to %Y-%m-%d**-%H**/%H-%M-%S - and I've been able to move a day and a half's photos into the folder...  takes a while to bring up the browser admittedly...

I'll see how things progress - if it works I don't mind the extra wait.

closed 2021-03-05 by starbasessd