motioneye-project/motioneye #1998: Motioneye Server/Hub crashed -> "An error occurred. Refreshing is recommended." most of the menu entries are gone

ovz93br43v7, 2021-01-15:
I added today my second camera to my motioneye server/hub and wanted to set up the motion detection.
After click on "Apply" motioneye told me "An error occurred. Refreshing is recommended."

Refresh didn't changed anything. At the moment both cameras are not linked to the motioneye server anymore. And the main menu only contains the points "preferences" and "general settings".
Normally I would check the log files via main menu to check some clear errors, but this menu entry is gone.

How can I find out want happened and how can I restore the menu, settings and linked cameras?

starbasessd, 2021-01-15:
Reboot the hub, then make sure you are logged in as 'admin'.
It sounds like it got logged in as user.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

ovz93br43v7, 2021-01-15:
Reboot I have already tried, and I'm logged in as admin, sorry I missed that.

ovz93br43v7, 2021-01-15:
As user I have no menu entries.

starbasessd, 2021-01-15:
> that to write.

What are you running? Which OS? What device (SBC or PC)? How did you
install?
There aren't any categories of settings for "preferences" or "General
Settings" in motioneye or motionEyeOS.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

ovz93br43v7, 2021-01-15:
RaspberryPi 4 with Raspbian
Install according to https://github.com/ccrisan/motioneye/wiki/Install-On-Raspbian

Ok I don't reemember how the menu was looking before I had added the two cameras. And I thought there is a at least the menu point with the log files, but this is also missing.
And everytime I log in I now see this:
[image]

ovz93br43v7, 2021-01-15:
I already refreshed and cleared cookies and website data

starbasessd, 2021-01-15:
re-image RaspberryPiOS
re-install motion/motionEye using the instructions for:
https://github.com/ccrisan/motioneye/wiki/%28Install-On-Ubuntu-%2820.04-or-Newer%29

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

ovz93br43v7, 2021-01-15:
Ok I will try that thanks

closed 2021-01-15 by ovz93br43v7

starbasessd, 2021-01-15:
The section with the links to the logs is only in Expert Settings in
motionEyeOS.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

ovz93br43v7, 2021-01-15:
I checked sudo systemctl status motioneye:

```
Jan 15 23:20:19 raspberrypi meyectl[388]:   File "/usr/lib/python2.7/shlex.py", line 96, in get_token
Jan 15 23:20:19 raspberrypi meyectl[388]:     raw = self.read_token()
Jan 15 23:20:19 raspberrypi meyectl[388]:   File "/usr/lib/python2.7/shlex.py", line 172, in read_token
Jan 15 23:20:19 raspberrypi meyectl[388]:     raise ValueError, "No closing quotation"
Jan 15 23:20:19 raspberrypi meyectl[388]: ValueError: No closing quotation
Jan 15 23:20:19 raspberrypi meyectl[388]:    ERROR: 500 GET /config/list/?_=1610749219319&_username=admin&_signature=6a08e3939a7ff5014b2
```

Does this help to narrow down the problem?

starbasessd, 2021-01-15:
Corrupted files.
Please follow my previous instructions.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

zagrim, 2021-01-16:
@ovz93br43v7 That log snippet does help to narrow things down indeed. There's something in the 2nd camera config which contains (likely) a quote (single or double) somewhere and MotionEye can't handle it. Probable places are a password or some token, or script or its parameters. If you have SSH connection to the server you could manually fix the condition either by removing the 2nd camera config altogether or by editing it to get rid of the offending quote character. Obvisouly you also need to also change the password or token or whatever so that things work without having the quote character in there.
The camera config file should be `/etc/motioneye/camera-2.conf`, and if you are willing to just throw it away, you'd probably better to also delete the reference to it from `/etc/motioneye/motion.conf` by removing the corresponding `camera` line in the file.

If you figure out in which setting the quote was, it would by helpful if you told it here so that the MotionEye UI validation could be improved to prevent the user for submitting something that the application can't handle.

ovz93br43v7, 2021-01-16:
@zagrim I already deleted all files from the motioneye server to make today or tomorrow a clean install. :-(

Regarding the possible special character which probably break something. In the beginning I used keepass generated random 128 character passwords with all special characters for user and admin passwords. (on camera 1) But I removed the special characters because I understood in my trials that this causes problems (looking on HDMI output during boot and saw errors and @starbasessd gave me the hint to the special characters in the passwords). Form this point on I removed every special character and used only passwords with 128 characters length [a-z][A-Z][0-9].

But nevertheless the passwords are still stored in my keepass, so I double-checked my passwords:

- [X] camera 1 user --> [a-z][A-Z][0-9]

- [X] camera 1 admin --> [a-z][A-Z][0-9]

- [X] camera 2 user --> [a-z][A-Z][0-9]

- [X] camera 2 admin --> [a-z][A-Z][0-9]

- [X] server user --> [a-z][A-Z][0-9]

- [X] server admin --> [a-z][A-Z][0-9]

My only password with special characters is my WiFi password, but the server is connected via LAN and WiFi is only used at the cameras.