motioneye-project/motioneye #2000: Accessing Motion Eye from Outside my Home using SWAG SSL

STUKguy, 2021-01-17:
Hi,
I am trying to set it so i can access my Motion Eye fro outside my home, now doing alot of back and forth on the OMV Forums i found the best approach for me is using my Nextcloud set up with SWAG & MariaDB set up, which all 3 run on the same network so i moved Motion eye over to this network but now i am stuck i no i need to locate a file called "motioneye.subdomain.conf.sample" and edit this but the file doesnt not exist.

Can someone guide me please?

starbasessd, 2021-01-17:
That is a HomeAssistant issue, please check on their forums.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

closed 2021-01-17 by starbasessd

STUKguy, 2021-01-17:
@starbasessd Its nothing to do with HomeAsstant!!! Im asking for help to create a file called "motioneye.subdomain.conf" in SWAG/NGINX and also eneything i need to change to be able to access **MOTIONEYE** from outside my home

starbasessd, 2021-01-17:
https://github.com/ccrisan/motioneye/wiki/Running-Behind-Nginx

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

STUKguy, 2021-01-17:
Yeah that didnt work either, Thank you for being so helpfull!!!