motioneye-project/motioneye #2002: Power Consumption for Solar

mroberts77, 2021-01-19:
Is it fair to assess that a Rasp Zero W running with the typical stock camera and specs below if using about 0.3Amp-hr? Even with the LED off, about the same?

Asking as I wanted to put off grid with solar but appears that my 2200mA LiPo isn't enough with a 5x500mA hrs solar charge to make it thru 1 day.

Can the motionOS system turn off/sleep during non-working hours to not use power (I.e. night)? If so, can it be done without unmounting or remote access only?

motionEye Version | 0.42.1
-- | --
Motion Version | 4.3.1
OS Version | motionEyeOS dev20201026

starbasessd, 2021-01-19:
There isn't anything built in to the Pis to be able to do so, nor into
motionEyeOS.
You could add a light sensor circuit to turn off the power to the Pi at
night and on during the day, and only the sensor circuit would need power
You could use a RTC to turn off the power to the Pi during specific hours,
but the RTC circuit would need power 24/7.
You could use an ESP device to listen to wifi for control signals to turn
the power to the PI on and off, they are known for their low power
consumption.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

jasaw, 2021-01-20:
@mroberts77 You could use an ESP32 camera module ($8 each I think) rather than a Pi. Set the ESP32 up as an IP camera. There are open source projects around that can get you started quickly.
Alternatively, if you really need to run a Pi, you'll have to increase your battery capacity and solar panel size. Remember to take into account of rainy days and short Winter days.

mroberts77, 2021-01-20:
On one part, I really like the interface of the Motioneye OS and web interface, controls and avoidance of the programming behind it: but for the purpose I have now as a timelapse, the motion detection isn't being leveraged and so could be a simpler set up that has a real time clock and takes pics on some sort of camera module. Will look into it, thanks for the advice.

closed 2021-01-20 by mroberts77