motioneye-project/motioneye #2003: EzViz camera FHD got error  "NOT MODULE 16"

Tavano, 2021-01-19:
Hi, I am asking for help regarding an installation of motioneye on Raspbian.
Installed the service, made it running properly, it is up to date.

I have configured only one camera (EzViz C3N FHD) and the problem is that:
1. The resolution (selected by the web interface menù) of 1920x1080 gives me error logs "NOT MODULO 16", but i can see the streaming correctly and the Motion detect is properly working.
2. The "Run A command" is not working at all (maybe caused by the error above..?), it is a mosquitto_pub to send the state to Home-Assistant. The command has been tested with a direct mosquitto_pub and is working properly.
3. Same problem for the END command

Thanks for anyone will help me!!

starbasessd, 2021-01-19:
2) Does the command used in Run a Command work if typed in a terminal window or at the command line on the Pi in question, or does it require 'sudo'?
3) Ditto for the End Command?
for #1, which instruction set to install motion/motionEye did you follow?

Tavano, 2021-01-19:
2. Mosquitto is installed on a docker container so, if I use the docker container shell, yes the command works perfectly, Here is the command:
mosquitto_pub -h IP_BROKER -u MQTTUSERNAME -P MQTTPASSWORD -t stat/motionEye/motion -m '{"ipcam":"on"}'
Where IP_BROKER is like 192.168.1.55 and is the IP of the raspberry that cotnains the motionEye and the docker container.

3. I mean that the Run End Command is not working, the end command is like the previous command:
mosquitto_pub -h IP_BROKER -u MQTTUSERNAME -P MQTTPASSWORD -t stat/motionEye/motion -m '{"ipcam":"off"}'

starbasessd, 2021-01-19:
Try changing the -t stat/... to the full path of the stat file.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Tavano, 2021-01-19:
Sorry, I not understood what you mean...
If it is the topic: -t stat/motionEye/motion
What are you suggesting to do?

starbasessd, 2021-01-19:
where is the file 'stat'?
If it is in /usr/bin/stat then you need to add the '/usr/bin/' before it in
the command
If it is somewhere else then you need to add that part of the path.
you can use the command
which stat [enter]
and it will show something like
/usr/bin/stat
so you would change it from "-t stat/..." to "-t /usr/bin/stat/..."
A lot of environments don't carry over the current path settings being
used, so you need to include the full path to the file/program...

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Tavano, 2021-01-19:
Ah ok, no. nothing changed. The camera app still alert me about motion detect. Also MotionEye, in the interfaces, detects the pixel change but no topic is published.

I have composed a command to lunch from the pi user (the default raspberry user)
sudo docker exec -it mosquitto mosquitto_pub -h IP_BROKER -u mqtt_user -P mqtt_password -t stat/motionEye/motion -m '{"ipcam":"off"}'

It works perfectly using the terminal but on motionEye no. Tryied with "sudo" and with the full path but nothing change. It won't publish the topic

Tavano, 2021-01-19:
[image]
Added the full path but nothing changed.
In the attached there is a screenshot with the commands you asked (putting "echo $PATH >> /home/pi/path.txt" in "Run A Command" give the same result that you can see in the screen shot) so it seems that the command is correctly executed

starbasessd, 2021-01-19:
Please try to answer all the questions posted.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Tavano, 2021-01-19:
Ok I'll try to answer...
Q: which instruction set to install motion/motionEye did you follow?
A: _https://indomus.it/guide/integrare-telecamere-come-sensori-di-movimento-su-home-assistant-via-motioneye/_

Q: Do you have motioneye service running as the Pi user?
A: _Using "ps auxw| grep meyectl" the output is:_
`root       352 10.3  0.2  63800 21008 ?        Ssl  17:48   0:20 /usr/bin/python /usr/local/bin/meyectl startserver -c /etc/motioneye/motioneye.conf
root       572  0.0  0.1  35100 12980 ?        S    17:48   0:00 /usr/bin/python /usr/local/bin/meyectl startserver -c /etc/motioneye/motioneye.conf
pi        2726  0.0  0.0   7348   476 pts/0    S+   17:51   0:00 grep --color=auto meyectl
`

Q: What does it show?
A: _Setting "Run a Command" as_ `echo $PATH >> /home/pi/path.txt` the file contains `/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin` and it is written periodically according to the motion detect

Q: Also do a which docker and add the full path to docker in the run command
A: Done and nothing changed

starbasessd, 2021-01-19:
Contact the HomeAssistant Community and FRENCK about any issues as that
version of motioneEye has been modded, but they won't say what all has been
changed.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Tavano, 2021-01-19:
May I install a different version of motioneye? Maybe a previous one to make mqtt work?

starbasessd, 2021-01-19:
You can try installing motion/motionEye using the following docker install or using the instructions here: 
https://github.com/ccrisan/motioneye/wiki/%28Install-On-Ubuntu-%2820.04-or-Newer%29

Tavano, 2021-01-19:
Thanks a lot!!!

Tavano, 2021-01-28:
Hi,
finally I solved the problem, here the steps:
1. Modified the file _/etc/pam.d/su_ added the following lines (to allow the execution of the script as "pi" user without asking pwd)

> \# This allows root to su without passwords (normal operation)
> auth       sufficient pam_rootok.so
> auth       [success=ignore default=1] pam_succeed_if.so user = pi
> auth        sufficient                              pam_succeed_if.so use_uid user = pi
> \# Uncomment this to force users to be a member of group root

2. created a script _camera_on.py_ for the motion occured

> #!/usr/bin/python
> import time
> import paho.mqtt.client as paho
> 
> #broker="broker.hivemq.com"
> broker="MQTT_BROKER_IP"   #default port 1883
> #define callback
> #def on_message(client, userdata, message):
> \#    time.sleep(1)
> \#    print("received message =",str(message.payload.decode("utf-8")))
> #
> client= paho.Client("client-001") #create client object client1.on_publish = on_publish #assign function to callback client1.connect(broker,port) #establish
> ######Bind function to callback
> #client.on_message=on_message
> #####
> #print("connecting to broker ",broker)
> client.username_pw_set(username="mqtt_user",password="mqtt_password")
> client.connect(broker)
> client.loop_start() #start loop to process received messages
> #print("subscribing ")
> client.subscribe("camera_alert/camera_garage")#subscribe
> time.sleep(2)
> #print("publishing ")
> client.publish("camera_alert/camera_garage","on")#publish
> time.sleep(4)
> client.disconnect() #disconnect
> client.loop_stop() #stop loop

2. Put the script in _/usr/local/bin/_
3. Made it executable _chmod +x camera_on.py_ 
5. Put in the WEB interface of motioneye the following "Run a Command"  _su -c camera_on.py pi_
6. Repeated steps 2 to 5 for the _camera_off.py_ script