motioneye-project/motioneye #2006: Failed to upload file with service gdrive

YuenTW, 2021-01-21:
Hi, i got some error message from Motioneye.log
ERROR: failed to upload file "/data/output/Camera1/2021-01-21/21-24-20.mp4" with service gdrive: 
Traceback (most recent call last):
  File "/usr/lib/python2.7/site-packages/motioneye/uploadservices.py", line 1103, in upload_media_file
  File "/usr/lib/python2.7/site-packages/motioneye/uploadservices.py", line 103, in upload_file
  File "/usr/lib/python2.7/site-packages/motioneye/uploadservices.py", line 371, in upload_data
MemoryError

not sure why its not working, it will auto upload yesterday, dont know why it stop uploading and shows up the error message.
gpu memory : 256
Movie format H.264
Moive Max length 3600s
i have reboot the pi but sill nothing positive message

Log : [motioneye.log](https://github.com/ccrisan/motioneye/files/5849944/motioneye.log)

starbasessd, 2021-01-21:
Versions of motion, motionEye and Base OS?
What install instructions were followed for motionEye?
Which version of pip2? [pip2 --version]
Can you access your GDrive from any other computer?

YuenTW, 2021-01-21:
MotionEye Version 0.42.1
Motion Version 4.2.2+gitUKNOWN
OS Version MotionEyeOS 20200606
Pip2 not sure, dont know how to check
Yes i could access GDrive from other device / pc

starbasessd, 2021-01-21:
Please close this issue, and open one at
https://github.com/ccrisan/motioneyeos/issues
Which Pi?
Please install /dev20201026 if Pi4, or 20190911 or dev20201026 if any other
version.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

closed 2021-01-21 by YuenTW