motioneye-project/motioneye #2008: Trying to install motioneye on fresh and updated ubuntu 20.04 VM following the instruction above but when I need to install pip2 (sudo python2 get-pip.py) I get an error:

mbrivio, 2021-01-25:
Trying to install motioneye on fresh and updated ubuntu 20.04 VM following the instruction above but when I need to install pip2 (sudo python2 get-pip.py) I get an error:
```
Traceback (most recent call last):
  File "get-pip.py", line 24226, in <module>
    main()
  File "get-pip.py", line 199, in main
    bootstrap(tmpdir=tmpdir)
  File "get-pip.py", line 82, in bootstrap
    from pip._internal.cli.main import main as pip_entry_point
  File "/tmp/tmpWF8rwr/pip.zip/pip/_internal/cli/main.py", line 60
    sys.stderr.write(f"ERROR: {exc}")
                                   ^
SyntaxError: invalid syntax
```

_Originally posted by @mbrivio in https://github.com/ccrisan/motioneye/issues/1069#issuecomment-766796893_

starbasessd, 2021-01-25:
> I've got the same issue!

They broke the upstream link, sorry.

New install for pip2 is
curl https://bootstrap.pypa.io/2.7/get-pip.py --output getpip.py
python2 getpip.py

Updating document.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

closed 2021-01-25 by mbrivio

jgrete, 2021-01-25:
Ptoblem fixed. Thank you!