motioneye-project/motioneye #2009: WARNING: Connect error on fd 14: ECONNREFUSED

popeyesfx, 2021-01-26:
Does anyone know how to fix this, the old thread was closed, but I dont see a fix for it.

starbasessd, 2021-01-26:
Which Base OS?
Which version of motion & motionEye?
Which installation instructions?
Which model board?
Which camera?
How is it connected? Ethernet/WiFi/USB/CSI?
Did it ever work with motioneEye?
Does the camera work in another OS?

popeyesfx, 2021-01-26:
The Base OS is debian buster for the pi zero
Motioneye was built using the docker file from the web site
the only difference is the use of : FROM balenalib/raspberry-pi-debian:latest : server says 0.42.1, motion is 4.1.1
The board is a pi zero version 2
Camera is the original pi camera 
Connected by Wifi to the network

Docker Run is
docker run --privileged --name="motioneye" \
    -p 8765:8765 \
    -p 8082:8081 \
    -p 7999:7999 \
    --hostname="motioneye" \
    -v /etc/localtime:/etc/localtime:ro \
    -v /home/pi/apps/motioneye/conf:/etc/motioneye \
    -v /home/pi/apps/motioneye/data:/var/lib/motioneye \
    --restart="always" \
    --detach=true \
    --device=/dev/video0 \
    popeyesfx/motioneye

What is strange is that the the container works and I can access the camera but the container log shows the error message
I have three of these camera's set up and each container shows anywhere from 2 to 5 warning messages

and I was wondering what might be the cause

thanks

starbasessd, 2021-01-26:
The Base OS is debian buster for the pi zero
Installed from where? 

     Motioneye was built using the docker file from the web site
There are 2 sets of instructions, which set?

the only difference is the use of : FROM balenalib/raspberry-pi-debian:latest : server says 0.42.1, motion is 4.1.1

     The board is a pi zero version 2
AFAIK, there isn't such an animal. V1.3 is latest PiZero without wifi, V1.1 is with wifi
Version is printed on the back of the Zero just below the GPIO pins

What else are you running on the PiZero? Any particular reason you are re-directing port 8081?

popeyesfx, 2021-01-27:
Sorry for not being a bit more specific, I am just trying to get an idea what might be causing the warning.

The pi zero's are part of my camera network for watching my boat. They are part of a docker swarm network which uses a raspberry pi 3b+ Rev 1.2 for the controller. This controller runs the standard motioneye container from docker hub. The Hub also runs a vpn container to allow me too remote into the network from home (boat is in Gloucestershire).
Each of the camera's uses a pi zero w rev 1.3 running docker and a motioneye container. (hence the diferent ports)

The containers are managed via portainer running on my local machine.

To build the container image
I changed the base image for one that will run on the pi zero,  balenalib/raspberry-pi-debian:latest ), in the docker file.
This was compiled using pycharm, setup with a remote connection to the docker host. (i use this setup all the time and it works great).
I have also compiled it using docker buildx on the local machine with the same results.

Everything works fine, just get those warnings and was wandering if they are caused my motion or motioneye or something else. Maybe a mismatch in the files.

starbasessd, 2021-01-27:
As stated before, the PiZero V1.3 is NOT wifi capable, how are you
connecting to it?
On the PiZeros (if, they are really V1.1 RpiZeroWs), why not just run
motionEyeOS in Fast Network Camera Mode? Lets the hub run all the 'heavy
lifting', and the Zero would be optimized to convert the video to the
stream. Just saying...

I cannot help support your setup as-is, unfortunately (modified docker on
PiZeros and Hub), , so hopefully someone will be able to help you.

popeyesfx, 2021-01-29:
I run docker containers as they are easy to manage remotely and i can change them without messing up the underlying system. so motioneye os isnt a solution.

The pi's are v1.1 zero's, I have so many I get mixed up sometimes.

I take it you dont know what FD 14 is then. Its weird I cant find anything on google other than stuff relating to motineye.

starbasessd, 2021-01-29:
What links I find googling " WARNING: Connect error on fd 14: ECONNREFUSED
" are related to motion-project, not motionEye. motionEye is a front-end
for motion and all the other programs.
Have you posted anything to the motion-project support?
https://github.com/Motion-Project/motion#resources

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

closed 2021-03-05 by starbasessd