motioneye-project/motioneye #2010: motionEye and Nginx ( {"error": "internal server error"} )

DooMFeaR, 2021-01-26:
Hello,
 Running MotionEye on a Raspberry Pi 3 B+ with Raspbian.

  After few hours, I get a blank page with the error message: 
`{"error": "internal server error"}`

If I reboot the raspberry pi, it works fine for few hours and then again same problem.

Is this an MotionEye or Nginx problem? Can someone help me debugging the reason of the error?

Thank you!

starbasessd, 2021-01-26:
Any log errors? 
Do you have all logging enabled in nginx?
https://docs.nginx.com/nginx/admin-guide/monitoring/logging/
If you connect direct to the 8765 port, is it still up and running?
If not, how about the motion & motioneye and messages logs?

DooMFeaR, 2021-01-28:
Same error on port 8765.

Motion.log 
`[0:motion] [NTC] [ALL] conf_load: Processing thread 0 - config file /etc/motioneye/motion.conf
[0:motion] [NTC] [ALL] config_camera: Processing camera config file camera-1.conf
[0:motion] [NTC] [ALL] config_camera: Processing camera config file camera-2.conf
[0:motion] [NTC] [ALL] motion_startup: Logging to syslog
[0:motion] [NTC] [ALL] motion_startup: Motion 4.2.2 Started
[0:motion] [NTC] [ALL] motion_startup: Using default log type (ALL)
[0:motion] [NTC] [ALL] motion_startup: Using log type (ALL) log level (WRN)`

messages
`Jan 27 23:17:14 pi3 motion: [0:motion] [NTC] [ALL] conf_load: Processing thread 0 - config file /etc/motioneye/motion.conf
Jan 27 23:17:14 pi3 motion: [0:motion] [NTC] [ALL] config_camera: Processing camera config file camera-1.conf
Jan 27 23:17:14 pi3 motion: [0:motion] [NTC] [ALL] config_camera: Processing camera config file camera-2.conf
Jan 27 23:17:14 pi3 motion: [0:motion] [NTC] [ALL] motion_startup: Logging to syslog
Jan 27 23:17:14 pi3 motion: [0:motion] [NTC] [ALL] motion_startup: Motion 4.2.2 Started
Jan 27 23:17:14 pi3 motion: [0:motion] [NTC] [ALL] motion_startup: Using default log type (ALL)
Jan 27 23:17:14 pi3 motion: [0:motion] [NTC] [ALL] motion_startup: Using log type (ALL) log level (WRN)
Jan 28 19:12:22 pi3 rsyslogd:  [origin software="rsyslogd" swVersion="8.1901.0" x-pid="388" x-info="https://www.rsyslog.com"] rsyslogd was HUPed`

starbasessd, 2021-01-28:
I should have been more specific, sorry.
Turn off nginx before testing port 8765 before testing, in case it's
intercepting that port...

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

DooMFeaR, 2021-01-31:
Same error buddy. So it means it's not an nginx problem!

starbasessd, 2021-01-31:
After a reboot, and the error re-appears, Console or SSH in and run
vcgencmd get_throttled
and report results.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

DooMFeaR, 2021-02-01:
vcgencmd get_throttled

`throttled=0x50005`

starbasessd, 2021-02-01:
Your power supply isn't up to snuff:

The bits on the returned number mean:

Bit Hex value   Meaning
0          1    Under-voltage detected
1          2    Arm frequency capped
2          4    Currently throttled
3          8    Soft temperature limit active
16     10000    Under-voltage has occurred
17     20000    Arm frequency capping has occurred
18     40000    Throttling has occurred
19     80000    Soft temperature limit has occurred

so yours is showing bits 1, 4, 16. & 18.

Need to look for a better power supply.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

DooMFeaR, 2021-02-01:
Weird, cos I'm using an 5V 10A power supply. This problem started a month or two ago. I'm going to try to check all the connections and see if I have any loose wires. Thank you for all your help.

[image]

closed 2021-03-05 by starbasessd