motioneye-project/motioneye #2011: Motioneye detects motion but doesn't save images for some camera

maxmonz69, 2021-01-26:
Hi,

I'm using motioneye 0.42.1 on a raspberry 2 with 3 different models of Vstarcam cameras. 
The number of cameras is 5 and everything works fine. I mean Motioneye detects motion on every camera and saves images and videos.

I recently tried the docker image and I reconfigured the system on a docker container on my NAS (importing the same configuration files used on raspberry). Everything works for 3 cameras (2 different models) but for two cams (same model), although the detection is working (I see the red borders around the changed area),  images are not saved.
If I take a snapshot manually it works. 
Any idea?
Is it related to the camera model? I can't imagine how. They are configured as generic network cameras by the streaming url which has the same format for each camera. So why, if motion and motioneye show the video stream are not able to capture images automatically?

Thanks

starbasessd, 2021-01-26:
Which docker install did you use [1] or [2]?
What is your base OS? 
What command do you use to start the docker session?

maxmonz69, 2021-01-26:
The docker version is 19.03.13, build feb6e8a9b5, installed by Docker Station on a QNAP NAS.
The OS is linux based but QNAP proprietary (QTS 4.5.1)
The docker image is ccrisan/motioneye:0.42-amd64.

The command to start the session is 
/bin/sh -c "test -e /etc/motioneye/motioneye.conf || cp /usr/share/motioneye/extra/motioneye.conf.sample /etc/motioneye/motioneye.conf ; /usr/local/bin/meyectl startserver -c /etc/motioneye/motioneye.conf"

starbasessd, 2021-01-26:
OK, I have NO experience running docker on a QNAP NAS, I can run docker in
a VM, in a linux or MAC machine, or on a Pi, which is what I was going to
try to do to reproduce your issue.
I'll leave this here if anyone else can help you.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

maxmonz69, 2021-01-27:
Thanks anyway.
The crazy thing for me is it works on my Pi2.
Soon I will try a new setup on a Raspberry Pi4. In the mean time, if someone has any idea, please write.

maxmonz69, 2021-02-12:
Hi,
there is an upodate. If I enable the debug mode, debug images are saved regularly for that camera. I'm getting crazy to understand why the regular images are not saved.
Any idea?

closed 2021-03-05 by starbasessd