motioneye-project/motioneye #2012: Pink Bands in live view. Blue 'negative' saved images

MyopicMonster, 2021-01-26:
Hello, Im having a persistent problem with both the Display Image and the Recorded (captured) Image. 
When viewing the live view, there is frequently vertical pink bars appearing in the view. Size, quantity, location on screen and frequency of the pink bars changes - no pattern but it does sometimes trigger a motion event capture. Sometimes the pink artifacts appear every few seconds on the screen, sometimes every few minutes. sample of image below.
Additionally, when a motion event creates a captured image, it is blue and black like looking at old-school 35mm negatives.
I have tried different camera resolutions, motion detection settings, , different (cheap Chinese) cameras and different 3A power supplies + cables without any change in behavior. Cameras tried have always been connected by ribbon cable-no USB cameras.
settings are below. advice greatly appreciated. Failing resolution, repurposing suggestions for the R-pi appreciated

 ;)

Device: R-Pi 4b 4G
motioneye: 0.42.1
Motion Version: 4.2.2+gitUNKNOWN
OS Version: motioneyeOS 20200606
camera: R-Pi V2 (NOT noir)
camera Device: vc.ril.camera
camera type: MMAL
resolution: 1280x1024
Frame rate Dimmer: 18
Resolution Dimmer: 85
file storage to attached USB
video streaming: OFF
Still Images-
Quality: 80%
Capture: motion triggered
Motion detection-
Frame Change: 5%
Max Change Threshold: 0
Auto Threshold: off
Auto Noise: off
Noise Level: 15%
Light Switch: 75%
Despeckle: on
Motion Gap: 20
Capture Before: 2
Capture After: 2
Minimum Motion Frames: 10
Mask: off
.
Motion Notification: off
Schedule: off
 
.
[bootJan26.log](https://github.com/ccrisan/motioneye/files/5876019/bootJan26.log)
.
[image]
.
.
[image]

starbasessd, 2021-01-26:
Please try dev20201026, there are some weird quirks with the Pi4 version of
20200606

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2021-01-26:
Yo hae abetter chance of getting pertinent answers for motionEyeOS issues
in the
https://github.com/ccrisan/motioneyeos/issues
area, this one is oriented to motionEye on another OS like Raspbian or
non-SBC hardware

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

MyopicMonster, 2021-01-26:
I have tried the pre-released file and it locked up the Pi device completely. the Pi was effectively bricked. It required a complete new set-up and re-write to the last released version 20200606.

MyopicMonster, 2021-01-26:
Relocating post to MotioneyeOS

closed 2021-01-26 by MyopicMonster

starbasessd, 2021-01-26:
I have one here in front of me running 4 cameras (3 MJPEG, 1 RTSP) with no
issues.
What kind of error(s) did you get? Did you hook a monitor to it to look?
What version of the boot loader is loaded?
vcgencmd  bootloader_version
If it's not dated later than Sep 2020 then you need to update it.
If you do a backup of your settings (Settings, General, Backup) you can
re-load it into 20200606 or dev20201026

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?