motioneye-project/motioneye #2014: Yoosee rtsp stream issue

ilmisu, 2021-01-29:
Hi everyone,
I bought a Chinese ip cam from Yosee, I can stream the flow from VLC correctly. From motioneye installed onto a PVE container, is not able to keep the streaming visible, is continuously blinking from grey to picture, also if I configure it with very low resolution and frame rate.
The container has got 4GB ram memory and 6 ipcamera attached.
Any hint? Thanks.

starbasessd, 2021-01-29:
User over on the motionEyeOS had an issue with the Yoosee camera, and solved it by picking rtsp & UDP when setting up the camera.
https://github.com/ccrisan/motioneyeos/issues/2624

ilmisu, 2021-01-29:
That is mandatory, Ive already see it and configured properly.

starbasessd, 2021-01-29:
I was never able to get motionEye to 'work' in the PVE environment. It does
work in a Docker Environment. You may want to ask on a PVE oriented forum,
or maybe someone here may be able to answer...

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

closed 2021-03-05 by starbasessd