motioneye-project/motioneye #2015: Upload on google drive

Sortberg, 2021-01-31:
Hi.
I use motioneye on a raspberry pi 4, and upload directly to google drive. I ka and barre not get my recordings to last longer than 19.sec. What do I get wrong? 
Thank you in advance 👍👍👍

starbasessd, 2021-01-31:
How long are the local files?

Sortberg, 2021-01-31:
> How long are the local files?

Actually, I don't know. I've only dat it up to google drive. What do you mean?

starbasessd, 2021-01-31:
Well, you can't ONLY upload to the GDrive. It creates a local file, then
uploads that to GDrive. If the local file is only 19 seconds, that
indicates a different issue, than if the local fiie 50 seconds and only the
GDrive versi is 19 seconds.
If you click on the camera video stream, a number of icons appear across
the top. One of them looks like a ">" which is the play button.
Click on it and it will start listing all the files locally. Click on one
and see if it is also 19 seconds.
If you see LOTs, make sure the local drive isn't full.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Sortberg, 2021-01-31:
Oh, I found it. Many thx. 
Is there the possibility to do so that it can take up longer than the approximately 20 seconds?

starbasessd, 2021-01-31:
It could be due to several settings:
Under Movies, Maximum Movie Length
Under Motion Detection, Motion Gap and Minimum Motion Frames
While testing, turn on Show Frame Changes.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Sortberg, 2021-01-31:
Thank you for your help. 

Last NOOB question. 

Do you know when I put Preserve Movies, for one of the days, after which it deletes the recording, then it is only the recording in motioneye, and not also the one in google drive?

starbasessd, 2021-01-31:
Theoretically Yes, but there is another setting that controls Cloud
deletions. and it doesn't work well..

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Sortberg, 2021-02-07:
> Theoretically Yes, but there is another setting that controls Cloud
> deletions. and it doesn't work well..
> 
> 
> 
> > Thank you for your help.
> >
> > Last NOOB question.
> >
> > Do you know when I put Preserve Movies, for one of the days, after which
> > it deletes the recording, then it is only the recording in motioneye, and
> > not also the one in google drive?
> >
> >
> 
> 
> Thanks
> 
> Kevin Shumaker
> 
> 
> Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>
> 
> N38° 19' 56.52"
> W85° 45' 8.56"
> 
> Semper Gumby
> “Don't tell people how to do things. Tell them what to do and let them
> surprise you with their results.” - G.S. Patton, Gen. USA
> Ethics are what we do when no one else is looking.
> Quis custodiet ipsos custodes?
> “There is no end to the good you can do if you don’t care who
> gets the credit.” - C Powell
> You know we're sitting on four million pounds of fuel, one nuclear weapon
> and a thing that has 270,000 moving parts built by the lowest bidder. Makes
> you feel good, doesn't it?

Maybe you can help me with one last problem?

As written I upload to google drive
However, the quality is poor when recording 
I can max run 800x600 in resolution. If I go over it there's a delay of 10 sec. How can I optimize it? I would like to run 1024x600. It's foscam 4K camera. 
Does changing gpu settings help in motioneye? Or is it my rpi 4 who can't keep up? 🙂

starbasessd, 2021-03-05:
Sorry, didn't see this last request until now.
I have a FosCam 9900. I use 1280x720 @ 10fps, with no issues. 
What URL are you using for the camera? You didn't mention the specific model...

JamBalaya56562, 2026-11-08:
Closing this, as the question was answered here: the file uploaded to Google Drive is the movie motionEye records locally, so its length is set by Movies > Maximum Movie Length and by Motion Detection > Motion Gap and Minimum Motion Frames, as starbasessd explained. Preserve Movies deletes the local files; uploaded ones are deleted along with them only when File Storage > Clean Cloud is on, which is available for Google Drive. If something is still unclear on a current release, please open a new issue.

closed 2026-11-08 by JamBalaya56562