motioneye-project/motioneye #2016: remote access outside the lan  problem ... any idea?

CrYpT0n1T3, 2021-02-01:
please help.
my settings not working
[image]
[image]
[image]
locally stream on port 8081 works - http: // 192.168.1.246: 8081
does not work from outside the network. http: // externalip: 8081 
netstat -a from motioneye
`Active Internet connections (servers and established)
Proto Recv-Q Send-Q Local Address           Foreign Address         State      
tcp        0      0 0.0.0.0:ssh             0.0.0.0:*               LISTEN     
tcp        0      0 localhost:7999          0.0.0.0:*               LISTEN     
tcp        0      0 0.0.0.0:www             0.0.0.0:*               LISTEN     
tcp        0      0 0.0.0.0:tproxy          0.0.0.0:*               LISTEN     
tcp        0      0 localhost:7999          localhost:58862         ESTABLISHED
tcp        0      0 192.168.1.246:ssh       192.168.1.176:33620     ESTABLISHED
tcp        0      0 localhost:40800         localhost:www           TIME_WAIT  
tcp        0      0 localhost:40792         localhost:www           TIME_WAIT  
tcp        0      0 localhost:40814         localhost:www           TIME_WAIT  
tcp        0      0 localhost:58862         localhost:7999          ESTABLISHED
tcp        0      0 localhost:40802         localhost:www           TIME_WAIT  
tcp        0      0 localhost:40808         localhost:www           TIME_WAIT  
tcp        0      0 localhost:40804         localhost:www           TIME_WAIT  
tcp        0      0 localhost:40796         localhost:www           TIME_WAIT  
tcp        0      0 localhost:40806         localhost:www           TIME_WAIT  
tcp        0      0 localhost:40798         localhost:www           TIME_WAIT  
tcp        0      0 localhost:40810         localhost:www           TIME_WAIT  
tcp        0      0 localhost:40794         localhost:www           TIME_WAIT  
tcp        0      0 localhost:40812         localhost:www           TIME_WAIT  
udp        0      0 0.0.0.0:bootpc          0.0.0.0:*                          
udp        0      0 192.168.1.246:ntp       0.0.0.0:*                          
udp        0      0 localhost:ntp           0.0.0.0:*                          
udp        0      0 0.0.0.0:ntp             0.0.0.0:*                          
udp6       0      0 fe80::65e:a4ff:fec3:ntp [::]:*                             
udp6       0      0 localhost:ntp           [::]:*                             
udp6       0      0 [::]:ntp                [::]:*                             
Active UNIX domain sockets (servers and established)
Proto RefCnt Flags       Type       State         I-Node   Path
unix  2      [ ]         DGRAM                    8250     /var/run/wpa_supplicant/wlan0
unix  7      [ ]         DGRAM                    6518     /dev/log
unix  2      [ ACC ]     SEQPACKET  LISTENING     6582     /run/udev/control
unix  2      [ ]         DGRAM                    18639    
unix  3      [ ]         DGRAM                    6592     
unix  2      [ ]         DGRAM                    8468     
unix  2      [ ]         DGRAM                    8503     
unix  2      [ ]         STREAM     CONNECTED     18063    
unix  3      [ ]         DGRAM                    6591     
unix  2      [ ]         DGRAM                    8507     
unix  2      [ ]         DGRAM                    8304     
`

starbasessd, 2021-02-01:
I would use something like
https://www.grc.com/
Shields Up to see if the firewall on your router is open from the outside
for the port forwarding of the 8081 port.
There are other websites out there, but this one has been there for years,
and is a little more trustworthy than some other random site...
It's possible that your ISP is blocking, or there's an issue with your
setup.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

CrYpT0n1T3, 2021-02-01:
ok
[image]

starbasessd, 2021-02-01:
BTW how did you test the outside? From a phone inside your house? If so,
make sure the wifi is off before the test.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

CrYpT0n1T3, 2021-02-01:
I typed on the phone with 3G
external router ip: 8081

starbasessd, 2021-02-01:
But was the wifi OFF on the phone?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

CrYpT0n1T3, 2021-02-01:
yes

starbasessd, 2021-02-01:
The GRC.com indicates the port is not open.

CrYpT0n1T3, 2021-02-01:
I also have ps4 and it detects double nat but everything works and I didn't have to set anything in the router.
So is it isp or dlink and firmware?

starbasessd, 2021-02-01:
PS4 uses an outside server and UPnP, motionEye does neither.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

CrYpT0n1T3, 2021-02-01:
I open stream locally on port 8081 and it works
netstat -a on motioneye doesn't show this port.

starbasessd, 2021-02-01:
try
sudo netstat -tulpn |grep LISTEN
on mine it shows:
kevin@camera-server:~$ sudo netstat -tulpn | grep LISTEN
tcp        0      0 0.0.0.0:8081            0.0.0.0:*               LISTEN
     11633/motion
tcp        0      0 0.0.0.0:8082            0.0.0.0:*               LISTEN
     11633/motion
tcp        0      0 0.0.0.0:8083            0.0.0.0:*               LISTEN
     11633/motion
tcp        0      0 0.0.0.0:8084            0.0.0.0:*               LISTEN
     11633/motion
tcp        0      0 0.0.0.0:8085            0.0.0.0:*               LISTEN
     11633/motion
tcp        0      0 127.0.0.53:53           0.0.0.0:*               LISTEN
     524/systemd-resolve
tcp        0      0 0.0.0.0:22              0.0.0.0:*               LISTEN
     2092/sshd: /usr/sbi
tcp        0      0 127.0.0.1:631           0.0.0.0:*               LISTEN
     1847610/cupsd
tcp        0      0 0.0.0.0:8765            0.0.0.0:*               LISTEN
     14464/python2
tcp        0      0 0.0.0.0:7999            0.0.0.0:*               LISTEN
     11633/motion
tcp6       0      0 :::22                   :::*                    LISTEN
     2092/sshd: /usr/sbi
tcp6       0      0 ::1:631                 :::*                    LISTEN
     1847610/cupsd

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

CrYpT0n1T3, 2021-02-01:
tcp        0      0 0.0.0.0:8081            0.0.0.0:*               LISTEN

starbasessd, 2021-02-01:
Listening on all addresses. It's a router issue from outside to inside.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

CrYpT0n1T3, 2021-02-01:
I called provider and he says they are not blocking ports

starbasessd, 2021-02-01:
Then ask that person how to set up your router.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

CrYpT0n1T3, 2021-02-01:
i am using my dlink router in bridge mode.
they won't help me with that.

starbasessd, 2021-02-01:
We can't, either.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

closed 2021-02-01 by starbasessd