motioneye-project/motioneye #2017: Store video files using AV1

matebuteler, 2021-02-02:
Hi. I really enjoy using MotionEye, but there's a problem: the video feed is encoded in H.264, and i have really low capacity hard drives. As the AV1 codec is open source, and ffmpeg supports it (https://trac.ffmpeg.org/wiki/Encode/AV1), is there any chance that i can store video using AV1?

starbasessd, 2021-02-03:
Might be better to ask on the Motion support:
https://github.com/Motion-Project/motion#resources

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

matebuteler, 2021-02-03:
Thank you! I'm sorry for asking here and not adding the tags- i don't ussualy use GitHub.

JamBalaya56562, 2026-11-08:
Closing, as the movies are encoded by Motion, so a new codec has to come from Motion first, as starbasessd said. For smaller files today, the movie format can be set to HEVC (.mp4), which takes less space than H.264 at the same quality.

closed 2026-11-08 by JamBalaya56562