motioneye-project/motioneye #2018: Follow movement PTZ

fjramirez1987, 2021-02-03:
Is it possible to obtain a value (x, y) of the area where motion is detected? I wanted to use that value to move my PTZ camera to the area where motion is detected.

Follow movement.

starbasessd, 2021-02-03:
You may also want to ask the support over at Motion-Project:
https://github.com/Motion-Project/motion#resources

otte845, 2021-02-23:
You can do that with the track_generic_move option in the .conf file [https://motion-project.github.io/4.3.1/motion_config.html#track_generic_move](url)
It's pretty easy to do once you find the commands to move your specific camera

JamBalaya56562, 2026-11-10:
Closing, as this was answered: Motion's tracking options do this, `track_type 9` with `track_generic_move` runs a command with the position of the motion, which can be put under Extra Motion Options and call the camera's own PTZ commands.

closed 2026-11-10 by JamBalaya56562