motioneye-project/motioneye #2019: Restoring default motion.conf values

anon-tigress, 2021-02-03:
Through an act of extreme carelessness I accidentally deleted my /data/etc/motion.conf file after spending hours configuring it. Now I cannot access the motioneye web interface, but curiously I can still ssh into the device running motion. Is there anyway to restore the motion.conf values to the default settings? I could not find the default values listed anywhere. I saw a suggestion to apt-get motion, but the problem is I used a direct image of motionOS and did not use apt-get to install it. If I try to run apt-get it says command not found.

starbasessd, 2021-02-03:
You could create another SDCard and copy the file from there to a local PC,
then re-insert the oops'd card and copy it on there.
Checking to see if there is a motion.conf.sample' anywhere else...

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

anon-tigress, 2021-02-03:
Good news, I was able to find that someone had posted the default settings in #1623 

Listed as: 
```
# @enabled on
# @show_advanced on
# @normal_password 
# @admin_username admin
# @admin_password 
# @normal_username user

webcontrol_port 7999
setup_mode off
webcontrol_parms 2
webcontrol_localhost on

camera camera-1.conf

webcontrol_interface 1
```

Do these values look ok?

zagrim, 2021-02-03:
Looks exactly like the copy in my setup (except for passwords and camera config references, of course). Just add more `camera` lines there if you have multiple cameras.

closed 2021-02-03 by anon-tigress