motioneye-project/motioneye #2021: Added 2nd camera and only shows 1 FPS counter overlay

sirweazel, 2021-02-03:
# Problem/Motivation

> (Why the issue was filed)
I added a 2nd camera, but the Streaming/Capture FPS counter is only showing 1 FPS counter.  So i don't know if its showing me the streaming or the Capture counter

## Expected behavior

> (What you expected to happen)
I should see 2 counters like the first camera that is added.

## Actual behavior

> (What actually happened)
Only see 1 counter

## Steps to reproduce

> (How can someone else make/see it happen)
Currently running Motioneye 11.0 via Home Assistant, but i once ran motioneyeos and i think i saw this same issue.  I am running different models of IP cameras via rtsp streams.  The first camera was a cheap IP camera, and the 2nd camera i'm adding is an Amcrest IP camera.  Let me know what other information you may need.

Sorry if i'm missing something, i'm probably just overlooking a configuration maybe?
Also note: i opened an issue under hassio-addons and i will update where needed.  I'm thinking it's me probably not understanding something and just overlooking something.

Screenshot attached:
[image]

starbasessd, 2021-02-03:
Which version of motion & motionEye? (In Settings (port 8765 WebGUI))
What is your base OS?
Your best support for this issue is going to be the HA Community, & FRENCK, as they forked the motionEye to work with HomeAssistant.
I am running many versions of motionEye on other OSs, and several versions of motionEyeOS on Raspberry Pis (all models) and none have this issue.

sirweazel, 2021-02-03:
Running under Hassio QCOW2 KVM image from thier downoads.

Motioneye version under Motioneye settings:
motionEye Version | 0.42.1
Motion Version | 4.3.2
OS Version | Linux 5.4.86

Hassio MotionEye version under Supervisor: Current version: 0.11.0
Hassio Version Info:
core-2021.1.5
supervisor-2021.01.7

So you don't think i'm missing a setting somewhere?  If one counter is showing, then they both should be showing? I know i messed with all the settings under my first camera to try and duplicate the opposite way to showing only 1 counter, and I was unable to.

starbasessd, 2021-02-03:
I don't know about the Hassio version, as they won't discuss their changes.
They are using a newer version of Motion.
I've never been able to get only one of the 2 frame rates to show, all the
way through latest motionEye or motionEyeOS dev20201026.
I am running Motion 4.2.2 on my motionEye production server due to other
issues with motion/motionEye  interaction.
If you get a response from FRENCK or the HomeAssistant Community, let us
know, please.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

sirweazel, 2021-02-04:
@starbasessd 
Here is a link to my home assistant github issue. Can you give it a review and let me know how to proceed. When I have a little extra time try and replicate on other environments I will (another Linux distro, it motioneyeos).  But, in the meantime if there is other info, logs, ect that I can provide our test I will.

Link to hassio issue I opened.
https://github.com/hassio-addons/addon-motioneye/issues/167

I'm on my phone, and it would let me link with the proper link syntax for some reason.

starbasessd, 2021-02-04:
> hassio-addons/addon-motioneye#167

Thank you, but no.
Until FRENCK and the HA Community are more open with the changes they make
to other's software (like motionEye and other packages), I am not
interested in helping them resolve their user's issues.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2021-02-04:
I would do as you suggested. Spin up a VM, add motion/motionEye using the
instructions for Ubuntu 20.04 and newer, (due to the upstream issues with
Python2.7) and see if it occurs there. If you have a spare SDCard, I'd do
the same for motionEyeOS dev20201026 on a Pi4, or 20190911 on any other PI,
too.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

sirweazel, 2021-02-04:
Thank you, sounds like a plan.  Next "free" day i have I'll try and replicate in other environments to try and track down the issue then proceed from there. I'll be silent for a little while as i don't see me having much time in the next couple days.  But shouldn't be "too long".  Thanks again.

starbasessd, 2021-02-04:
BTW, for comparison, here's 5 of my cameras, 1 USB connected web cam, 3 d-Link DCS-5020Ls and a FosCam 9900, and all 5 show both FPS for stream & record on my production machine running Ubuntu 20.04 & motion/motionEye. Same on my Pis (Pi3B+ and 4B-4GB running any version of motionEyeOS from 20190911, 20200606, or dev20201026, or even Ubuntu 20.10 for RPi...
[image]

Clewsy, 2021-02-04:
Hi there,

> Thank you, but no. Until FRENCK and the HA Community are more open with the changes they make to other's software (like motionEye and other packages), I am not interested in helping them resolve their user's issues.
> […](#)

FYI it's literally open source, all hosted here on github.  @Frenck makes no changes at all to motioneye, it's just packaged up in a container to run with home assistant.

@sirweazel good luck, I'll follow along here to see how your testing goes.

Also, FYI, unable to replicate with my configuration - 4x IP cams and a direct connected usb cam.

starbasessd, 2021-02-04:
BTW, I consider wrappers that change behavior to be changes. MotionEye is a
wrapper, but doesn't change behavior, it adds the WebGUI interface. Motion
and FFMPEG can be talked to directly, which doesn't seem to be the case
inside FRENCKs container...
Did you steal my setup? [LoL]

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

frenck, 2021-02-05:
Wow, the pointing. Sounds all kinda hostile, to be honest. Nobody means any harm, as a matter of fact, its promoting the use of motionEye even.

Seems that that is not appreciated.

> as they forked the motionEye to work with HomeAssistant.

Maybe you should have taken a small moment to look upstream before throwing these things. It's not a fork, if so, I would have happily taken on the issue report myself.

> Until FRENCK and the HA Community are more open with the changes they make
to other's software (like motionEye and other packages), I am not
interested in helping them resolve their user's issues.

It is open source. Not 1 single piece is hidden (even full builds have always been public):

<https://github.com/hassio-addons/addon-motioneye/>

As for the concern on modifying, it doesn't do that. It runs a pip install. But, of course, you would have seen that if you actually took a look before jumping in hostile.

Sorry to hear people are having problems and sorry to hear you don't like people promoting your product (which they really like, with over 1M pulls on Dockerhub, around 20~25K active instances).

I would have happily helped and joined all kinds of development and searching for solutions. But not when this is the way this project is going to approach this for now and the future.

../Frenck

starbasessd, 2021-02-05:
My apologies, sir.
I am not a programmer, or a coder. I am a Senior Help Desk for corporate
Enterprises, supporting many applications and OSs.
My first attempts for assisting users of any package is to attempt to
reproduce the issue.
I have a decent VirtualBox VM environment, for OS based testing, Pi1, Pi2
Pi3B, Pi3B+, Pi4-4GB, Pi4-8GB, PiZero, Zero and ZeroWH to test with.
I regularly install / re-install in all devices and environments, and have
modified the install instructions for motionEye, and, am getting ready to
make other changes due to upstream Python2.7 and pip2 issues.
I have been learning Docker environments to be able to help users there,
too.
I don't intend to point fingers, just point users to better resources or
support than what I can or am willing to do.
If this were a non-volunteer position, I could and would devote more time
and effort to learn more environments.
I am sorry that I came across that way, and if you have a better contact
framework, I would be more than happy to send users of your wrapper to it.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

sirweazel, 2021-02-05:
You are both a lot of help, thank you. I'll get back, once i get some more testing done.

sirweazel, 2021-02-05:
oops, wrong button

sirweazel, 2021-02-27:
just posting that i haven't forgotten or that this topic died because i said i didn't think it would take me long to test, but this latest bout of weather and ice hit hard, and now i have to unmount my cameras from the building and bring them inside to see if they still work.  So taking me a little longer than i originally thought.

starbasessd, 2021-02-27:
No problem. Locally, we've had multiple times more than our regular amounts of snow, 3 ice storms, colder temps than we normally get, etc, over the past 3 weeks, so I can relate. Take your time and be safe, especially if you need to use ladders. Don't forget the ground may be saturated and they can sink in easily.

sirweazel, 2021-04-25:
Cameras are back to operational. Thank you for the safety warning.  This winter had a couple "gotcha's" lol.  I'm back to looking at this and getting my cameras back in order.  I was hoping after deleting and re-adding the cameras back, it might of changed.  I'm only a couple hours in (i tend to move slow) but i'm back to only having the one FPS counter. I added the camereas UDP stream this time around, but i can't remember if when i first setup i had TCP or UDP streams.  Maybe it's a camera setting.  Thanks for not closing the ticket automatically.  Anyways, now that i'm back to it, hopefully i'll find something out.  I was able to remote into the Home Assistant motioneye container to get to the configs and stuff if needed.  When i compared configs ealier, nothing stuck out as being different to my other camera stream.  I'm just beginning to wonder if it just has something to do with the cameras i'm using.  Anyways, i'll post back later.

starbasessd, 2021-04-25:
I tried enabling / disabling various features on different type cameras and
was not able to reproduce the issue on my production, nor any of my test
systems, including PC/Ubuntu,motionEye or Pi4 motioneyeOS dev20201026,
Pi4/motionEyeOS20200606, Pi4/RaspberryPiOS/motioneye, PC/Ubuntu/docker,
Pi3B+Raspberry PiOS/docker, etc, all connecting t 5 cameras of 3 different
types.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

sirweazel, 2021-04-25:
So I'm goofy... nothing new to me though.  I'll keep at it.  If there is a way, i'll break it, but then i usually find a way to fix it.  I'll just keep at it. Thanks for checking, it's much appreciated.

sirweazel, 2022-10-06:
I'm closing this. Thanks for all the help. I never really got the time to dig into it, but i did have multiple installs since i opened this issue and same thing.  It must be something with the cameras.  Anyways, i'm closing this out.  I've actually moved on to other camera software for my needs. But i might revisit this for other projects. Thank you.

closed 2022-10-06 by sirweazel