motioneye-project/motioneye #2022: Grey camera from docker setup with netcam.

Xharos, 2021-02-06:
Hi, love your work

I've build the latest dev version according to :
- https://github.com/ccrisan/motioneye/wiki/Install-In-Docker

I run it with :
- `docker run --name="motioneye" -p 8765:8765 ccrisan/motioneye:master-amd64`

Everything is working fine, I can access the webui through localhost:8765

I've a network camera streaming rtsp at the following url :
- rtsp://192.168.1.21:8554/stream 

with no authentification, the live feed is working great (I can access it via VLC for example).

The problem is when I add my net cam in motioneye, I keep getting the grey screen preview.

Here is all my setup information :

- [ ] docker desktop for windows
- [ ] image motioneye:master-amd64 build from dev branch with ` docker build --build-arg VCS_REF=$(git rev-parse HEAD) --build-arg BUILD_DATE=$(date +"%Y-%m-%dT%H:%M:%SZ") -t ccrisan/motioneye:master-amd64 -f extra/Dockerfile .` command

and some logs I was able to pull outside the container :
`WARNING: 403 GET /login/?_=1612605826923&_username=_&_signature=82082a53af2fea4721d8c0154a72d3e92f706220 (172.17.0.1) 0.46ms
Cannot open device /dev/video0, exiting.
    INFO: adding new netcam camera with id 1...
    INFO: main config file /etc/motioneye/motion.conf does not exist, using default values
   ERROR: mjpg client timed out receiving data for camera 1 on port 8081`

same result with TCP & UDP connection, the display say unable to open video device

Xharos, 2021-02-06:
`[1:ml1:Camera1] [ERR] [VID] vid_start: Netcam RTSP failed to open
[1:ml1:Camera1] [WRN] [ALL] motion_init: Could not fetch initial image from camera
[1:ml1:Camera1] [WRN] [ALL] motion_init: Motion continues using width and height from config file(s)
[1:ml1:Camera1] [WRN] [ALL] mlp_retry: Retrying until successful connection with camera
[1:ml1:Camera1] [ERR] [NET] netcam_rtsp_open_context: Normal resolution: Unable to open camera(Camera1): Protocol not supported`
var/log/motion.log

starbasessd, 2021-02-06:
I will be following up with thai, but I had a system failure on my primary
Win10 box.  It may be a day or so.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Xharos, 2021-02-08:
Hi,

In fact, I just would like to know if my camera setup (RPI0W, VLC, raspivid stream mjpeg) is detected by motioneye.
If it's easier to start a proper linux vm, I could do that.

If it's works, I know I can buy a proper NAS to handle all the motion detection stuff.

Xharos, 2021-02-15:
Hey,

Did you find the time to try this setup with your machine?

Tanks a lots

starbasessd, 2021-02-15:
I have a clean install of Win10 Pro with all updates, and WSL WSL2 and
needed ancillary products installed and updated.
I've installed Docker Desktop and checked updates, and installed Ubuntu
20.04 with all its updates.
I am unable to get motionEye docker installed under Docker Desktop.
I even have Docker installed in the Ubuntu 20.04 and motionEye working
there.
I have VMs for several flavors of Linux and dockers working there.
The command line for Docker Desktop is enough different that I have not
been able to surmount the learning curve, yet.
Thoughts on it?

<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virus-free.
www.avg.com
<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Xharos, 2021-02-16:
I'm using docker desktop through the classic windows cmd since the application install docker in PATH env variable.
I'm able to start container the same way on my mac or debian server.

Xharos, 2021-02-16:
I also made a test from my mac (also with docker), I'm experiencing the same issue with the same error log.
(latest docker version with motion eye build with the command above)

I think the problem came from my camera but I don't know why

starbasessd, 2021-02-16:
I am firing up my old test MAC (10.13.6 High Sierra) and see what damage I
can cause...

<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virus-free.
www.avg.com
<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2021-02-16:
Are you running Docker Desktop, or just Docker on your MAC?

<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virus-free.
www.avg.com
<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Xharos, 2021-02-16:
just docker

Xharos, 2021-02-16:
I was able to pull some new interesting logs that lead me to this issue :
https://github.com/Motion-Project/motion/issues/759

I had exactly the same problem. I'm now looking for this motion.conf to see if rtsp uses tcp is activated

starbasessd, 2021-02-16:
> Motion-Project/motion#759

Well, that went over like a lead balloon. My MAC is "late 2009", and Docker
doesn't start support until 2010... <sigh>

<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virus-free.
www.avg.com
<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Xharos, 2021-02-16:
Hey, some news.

I switched from VLC to mjpg streamer. It's heavier on the RPI 0W but the cam can handle full HD 30 fps and display it over a web server.

I try the cam in motioneye and it works like a charm, I was able to connect, change settings, everything worked perfectly.

I try to restart the container a few moments ago and I now see this :
[image]
all the sections have dissapear. I don't know how to get it back 😢 .

Funny, the same problem occured on my mac, starting the same container with the same command and all the settings are gone.

If you have any idea on how to get it back?

Thanks a lots
Val

starbasessd, 2021-02-16:
If you set a password before, and set a different one, or not set yet, (or
vice-versa) clear the cache in the browser, it holds onto the previous
password, fails to log in, and you get the user screen instead of the admin
screen.

<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virus-free.
www.avg.com
<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Xharos, 2021-02-16:
I clear my cache, but same problem.

I also try with firefox same problem

Xharos, 2021-03-14:
I'm still unable to see auth panel in docker, if you have any idea :/

> Hey, some news.
>
> I switched from VLC to mjpg streamer. It's heavier on the RPI 0W but the cam can handle full HD 30 fps and display it over a web server.
>
> I try the cam in motioneye and it works [...]

starbasessd, 2021-03-14:
I have been unable to reproduce the original issue, nor your newest issue, with any Docker setup or direct setup, in several OSs, on several hardware bases. I am leaving this here in case someone else has ideas.

acyanic, 2021-04-13:
Same/similar issue here. 
Two cameras attached to Motioneye running in Docker on a Thinkpad server:
Camera 1: Successful IP cam rtsp stream, with motion detection and saving via SMB to NAS.
Camera 2: Unsuccessful Raspberry Pi Zero W with rtsp stream, 'Unable to open video device'.  Detection off.
VLC can open the Pi stream.

Note, the IP cam requires a user name password, the Pi cam does not (so NULL fields).  Is that the issue?

Logs and set up files following:

These lines repeat over and over in the Docker logs motioneye (extract):
   DEBUG: received relayed event stop for motion camera id 2 (camera id 2)
   DEBUG: clearing motion detected for camera with id 2
   DEBUG: 200 POST /_relay_event/?_username=admin&event=stop&motion_camera_id=2&_signature= ... (127.0.0.1) 1.09ms
   ERROR: mjpg client timed out receiving data for camera 2 on port 8082
   DEBUG: motion detection is enabled for camera with id 1

Motioneye installed in Docker on a Thinkpad laptop
Version:           19.03.6  API version:       1.40
 Go version:        go1.12.17
 Git commit:        369ce74a3c
 Built:             Fri Dec 18 12:21:44 2020
 OS/Arch:           linux/amd64

Docker image :  ccrisan/motioneye:master-amd64
  motionEye Version | 0.42
  Motion Version | 4.2.2
  OS Version | Ubuntu 18.10

docker run --name="motioneye"    -p 58082:8082 -p 8765:8765 -p 58081:8081 -p 1424:1424   --hostname="motioneye"     -v /etc/localtime:/etc/localtime:ro     -v /etc/motioneye:/etc/motioneye     --mount src=nas-files,dst=/toNAS      --restart="always"     --detach=true     ccrisan/motioneye:master-amd64

Raspberry Zero W running DietPi V 7.0
running as non-root user
raspivid -o - -t 0 -n -w 1280 -h 720 -fps 20 -b 250000 | cvlc stream:///dev/stdin --h264-fps=20 --sout '#rtp{sdp=rtsp://:8554/stream}' :demux=h264

Camera1.conf  (IP cam that works)
# @webcam_resolution 100
# @upload_subfolders off
# @upload_server 192.168.1.9
# @enabled on
# @network_server 
# @motion_detection on
# @upload_port 1424
# @upload_location /Watching/cam2/
# @preserve_movies 30
# @network_username 
# @upload_movie on
# @id 1
# @webcam_server_resize off
# @upload_password ****
# @manual_record off
# @upload_method post
# @upload_picture off
# @working_schedule_type outside
# @network_password 
# @upload_service ftp
# @preserve_pictures 0
# @storage_device custom-path
# @manual_snapshots on
# @network_share_name 
# @upload_enabled off
# @network_smb_ver 1.0
# @working_schedule 08:00-20:00|08:00-20:00|08:00-20:00|08:00-20:00|08:00-20:00|08:00-20:00|08:00-20:00
# @clean_cloud_enabled off
# @upload_username ***

threshold_maximum 0
netcam_userpass **:**
stream_quality 85
threshold 1843
[... 114 lines cut ...]
minimum_motion_frames 20
stream_motion off
target_dir /toNAS/cam2/picam
movie_output_motion off
post_capture 1
stream_authentication motion:dothelocomotion
on_picture_save /usr/local/lib/python2.7/dist-packages/motioneye/scripts/relayevent.sh "/etc/motioneye/motioneye.conf" picture_save %t %f
on_movie_end /usr/local/lib/python2.7/dist-packages/motioneye/scripts/relayevent.sh "/etc/motioneye/motioneye.conf" movie_end %t %f
text_left Camera2
picture_output_motion off
picture_filename 
text_scale 1
locate_motion_style redbox
locate_motion_mode off
netcam_keepalive off
movie_quality 75
picture_output off
on_event_end /usr/local/lib/python2.7/dist-packages/motioneye/scripts/relayevent.sh "/etc/motioneye/motioneye.conf" stop %t
text_right %Y-%m-%d\n%T
on_event_start /usr/local/lib/python2.7/dist-packages/motioneye/scripts/relayevent.sh "/etc/motioneye/motioneye.conf" start %t
camera_name Camera2
netcam_tolerant_check on
height 720
mask_file 
width 1280

acyanic, 2021-04-13:
how about in the code, does it need to allow null values for use and password .... in the call to the rtsp stream

utils.py
about line 444
def test_rtsp_url(data, callback)
...
   username = data.get('username')
    password = data.get('password')

whereas a few lines above in def 'response for the mjpeg code'
   username = data.get('username') or None
    password = data.get('password') or None

acyanic, 2021-04-17:
it's not user/pass issue.
I used '#rtp{sdp=rtsp://:8554/stream/,user='a',passwd='a'}'
No change to VLC working and motioneye not, (both prompted for password)

acyanic, 2021-04-25:
ok, solved for me.
I've played and played with settings in the command line.  Running vlc and FFmpeg players, ffprobe, log files all over the place. Most times VLC and Fffmpeg would play the stream but motioned would not.

These errors occurred during the journey:
    non-existing PPS 0 referenced
    decode_slice_header error
Referenced here in [stackexchange](https://stackoverflow.com/questions/15005436/errors-when-decode-h-264-frames-using-ffmpeg)

Checking raspivid command parameters, inline headers are required to repeat the keyframe info.
   Raspivid option --inline solved the problem.

My command line now :
`raspivid -o - --inline -t 0 -n -w 1280 -h 720 -b 2000000 | 
cvlc -I dummy stream:///dev/stdin --sout '#rtp{sdp=rtsp://:8554/stream}' :demux=h26x --lua false`

MotionEye running in a docker container, with motion detected files written to NAS.

Note that the lua option for vlc doesn't seem to disable all the meta data lookups, not a big deal just unnecessary noise in the log files.

acyanic, 2021-04-25:
and a new error after cam running a while, ES_OUT_SET_(GROUP_)PCR
apparently this [stackexchange](https://raspberrypi.stackexchange.com/questions/95420/raspivid-es-out-set-group-pcr-is-called-too-late-stream-keeps-freezing#103808) answer

add specific fps options to raspivid and vlc

`raspivid -o - --inline -t 0 -n -w 1280 -h 720 -fps 20 -b 2000000 | 
cvlc -I dummy stream:///dev/stdin --sout '#rtp{sdp=rtsp://:8554/stream}' :demux=h26x --lua false --h264-fps=20`

JamBalaya56562, 2026-11-08:
Closing, as both setups here were solved on the camera side: the stream of the Pi Zero W worked once it was served by mjpg-streamer instead of VLC, and acyanic's once raspivid and VLC got fixed frame rate options. "Netcam RTSP failed to open" with a stream VLC can play comes from Motion, as in Motion-Project/motion#759. The missing settings after restarting the container are a separate matter; please open a new issue if that still happens with a current image.

closed 2026-11-08 by JamBalaya56562