motioneye-project/motioneye #2023: How can I setting the Mmal camera that is connected to the PI with a ribbon cable in the home assistant platform? ?

massimoleonardi, 2021-02-06:
I can try to connect PI camera   that  is connected to the PI with a ribbon cable adding command in configuration.yaml  with the only line rpi_camera:   but is not recognize in UI and in motionEye ( in home assistant )

starbasessd, 2021-02-06:
It should be auto-discovered in motion/motionEye without using HA/wrapper.
If you connect to port 8765 to gett the motionEye screen, if it doesn't
show there, the odds are that the cables are facing the wrong direction,
either on the camera or the PI.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

massimoleonardi, 2021-02-07:
unfortunately is not working. the same Raspberry Pi ( with different SD ) on OS platform and camera , working well . The issue is only in HA . So I can not see any camera in the mail panel or in the motionEye. Are any command in the  configuration.yaml that I have to add ?

starbasessd, 2021-02-07:
I do not have much experience with HA. Have you asked in their Community
Forums?
If you install motion/motionEye without HA on that other SDCard, does it
work?
Use the instructions from
https://github.com/ccrisan/motioneye/wiki/%28Install-On-Ubuntu-%2820.04-or-Newer%29
due to issues with python2.7 and upstream sources for pip2

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

JamBalaya56562, 2026-11-08:
Closing, as this was about the MMAL camera in the Home Assistant add-on of 2021, and the same Pi worked without Home Assistant. MMAL is the legacy camera stack, removed from Raspberry Pi OS since Bookworm; current Pi cameras go through libcamera, see [Troubleshooting](https://github.com/motioneye-project/motioneye/wiki/Troubleshooting#libcamera-camera-stack). The add-on is maintained by the Home Assistant community add-ons project.

closed 2026-11-08 by JamBalaya56562