motioneye-project/motioneye #2024: docker port 8083 not working

myB0B, 2021-02-07:
trying motioneye in docker and port 8083 is not working

error message in console :
 WARNING: mjpg client connection for camera 3 on port 8083 is closed

i force opening port with -p 8083:8083 but nothing more
port 8081, 8082, 8084 etc... are working well

starbasessd, 2021-02-07:
Would you please post your original docker command and/or your docker run command? Copy/Paste, please, looking for a syntax / wrong character issue. 
I have both docker install & build on 2 different VMs, with 5 cameras on each, and all are working on both.

myB0B, 2021-02-07:
Docker Desktop on windows server 2019
docker compose :

  motioneye:
    container_name: motioneye
    image: ccrisan/motioneye:master-amd64
    restart: always
    ports:
      - "8765:8765"
      - "8081:8081"
      - "8082:8082"
      - "8083:8083"
      - "8084:8084"
    volumes:
      - E:\VM\DOCKER\_config\motioneye:/etc/motioneye
      - E:\VM\DOCKER\_config\motioneye:/var/lib/motioneye

no problem with other ports, only 8083 with message :
WARNING: mjpg client connection for camera 3 on port 8083 is closed

starbasessd, 2021-02-07:
If you run
netstat -an |find /i “*listening*”
on WinServer2019 what ports in the 8080-8090 range show?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

myB0B, 2021-02-07:
not a ports problem on the host machine
i have a VM with debian10 and motioneye running well on the same machine
problem only appear with the docker image

starbasessd, 2021-02-07:
If I block a port on a WinServer2016 VM doesn't show on the Ubuntu Host or
any other VM running on the host.
Also, if something else (like a previous docker motionEye)) has control of
the port on that server you won't be able to use it on that server
I am spinning up a 2019 Server VM right now.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

myB0B, 2021-02-07:
there is no other docker images...

just try many other ports from 8084 to 8094 and all are working well
also try NAT form 8083:8093 and this is working well !
so this is a internal problem inside the docker image with port 8083

starbasessd, 2021-02-07:
Still spinning up the Server2019, then will install Docker desktop, and do
both Install Docker Image and Build Docker, and will post my results.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2021-02-07:
Got my Server 2019 installed and fully updated,
Installed Docker Desktop 3.1.0
but every time I try to start it, I get an error. Tried with Hyper-V
selected and not selected.
If you have any ideas on getting Docker to run and accept commands, I'd be
happy to try them.
On my other machines with docker, the motionEye Docker works fine. I have 5
cameras installed in all versions, all respond to 8081-8085.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

myB0B, 2021-02-08:
you have to activate nested virtualisation (if you can) if you want to run Docker Desktop in a VM. 
https://docs.microsoft.com/en-us/virtualization/hyper-v-on-windows/user-guide/nested-virtualization

starbasessd, 2021-02-08:
Oopps. Interesting. My Host is AMD Ryzen 7 / Ubuntu20.04, so I guess it's not going to work for me. Sorry. Someone else will have to try to reproduce in that environment.

myB0B, 2021-02-09:
yes, only intel compatible

starbasessd, 2021-02-09:
OK, 'borrowed" a grandson's i5-Win10 Pro machine, installed WSL & WSL2 & Ubuntu 20.04. 
Installed Docker Desktop 3.1.0. All fully updated.
Got everything but the 808x ports set up and working. What am I missing? It didn't allow me to change/add other ports.

starbasessd, 2021-02-09:
BTW, I really want to understand this version of docker, which is why I am continuing to try to see if this is an issue.

starbasessd, 2021-02-09:
OK. This was 'interesting'.
Using command line:
docker run --name="motioneye" -p 8765:8765 -p 8081:8081 -p 8082:8082 -p 8083:8083 -p 8084:8084 --hostname="motioneye" -v C:\VM\DOCKER_config\motioneye:/etc/motioneye -v C:\VM\DOCKER_config\motioneye:/var/lib/motioneye --restart="always" --detach=true  ccrisan/motioneye:master-amd64
(all single line)
I got:
[image]
I then ran http://localhost:8765 in Edge
I got the WebGUI interface
I added 4 cameras (3 d-Link DCS-5020L & FosCam 9900).
I then connected to each of the 4 cameras in Edge tabs using http://localhost:8081 through 8084, all with no issues.
Unable to reproduce your issue, sorry.
Short of installing Server 2019, anything else I can try to match your setup closer?

starbasessd, 2021-02-09:
restarted with p 8085:8085 added, added a stream from my production motionEye Hub, and again, no issues, so 5 cameras is not a magic number, either...
I still vote for something mis-typed in the setup of the docker container, unless you or someone else can suggest something else to test...

starbasessd, 2021-02-09:
I ran 
docker inspect motioneye
scrolled down to "NetworkSettings"
[image]

myB0B, 2021-02-09:
damned....trying to change camera, others ports, still not working on port 8083
in parallel i have a wm with debian10 and 4 cameras running and port 8083 work fine
i dont know how to investigate more... i can't reinstall server, it is in prod
no problem for me, i use another port, but still a strange and incomprehensible thing

starbasessd, 2021-02-09:
You could try another instance of motioneye.
Tell docker to stop your current instance.
run the test image:
docker run --name="motioneye-test" -p 8765:8765 -p 8081:8081 -p 8082:8082
-p 8083:8083 -p 8084:8084 -p 8085:8085 --hostname="motioneye-test" -v
E:\VM\DOCKER_config\motioneye:/etc/motioneye -v
E:\VM\DOCKER_config\motioneye:/var/lib/motioneye --restart="always"
--detach=true  ccrisan/motioneye:master-amd64
This should bring up a new instance, with the same ports, and the same
volume paths.
Then check for active ports, and use the command
docker inspect motioneye-test
and see what the ports look like...

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

myB0B, 2021-02-09:
of course, try this many time with same result

starbasessd, 2021-02-09:
Then something is either specifically blocking 8083 for you, or has grabbed
control of it...

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

myB0B, 2021-02-09:
last but not least, i run a node-red docker image with changing port 1880 to port 8083  ( -p 8083:8083) both in node-red settings.js and host port, and it work ! so still sure this in a problem inside the motion image.
i don't kown how to invertigate inside the image and how the create one... 
BTW if someone can update version of motioneye inside the image

starbasessd, 2021-02-09:
Instructions in Part 2 of the Docker instructions:
https://github.com/ccrisan/motioneye/wiki/Install-In-Docker
Build Instructions.

Open the image to 'look around':
docker exec -it motioneye bash

Can you update inside the image? Nope.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

cap9qd, 2021-02-09:
You have a docker container for node red started with -p 8083:8083? Is this
running on the same host as the motioneye docker? If so then this is
probably what is blocking port 8083 on the motioneye docker from binding
internal port 8083 to host port 8083.

Is the node red contain starting first? It would be a good test to stop the
node red container and restart the motioneye one and see if it works.

cap9qd, 2021-02-09:
Or just move the node red to a different port on the docker host.

myB0B, 2021-02-11:
@cap9qd you did not understand what i wrote... just trying another docker image with port 8083 to test if non working port is bock by host or not

cap9qd, 2021-02-11:
Ah it was not clear in your message it was a test.  Sorry about that. Good
luck.

JamBalaya56562, 2026-11-08:
Closing, as this was with the `ccrisan/motioneye` image of 2021 on Docker Desktop for Windows, and you worked around it with another port. "mjpg client connection for camera 3 on port 8083 is closed" means motionEye lost Motion's stream of camera 3 inside the container, so publishing the port on the host could not change it. Please open a new issue with `motion.log` if one camera's port still fails with a current image.

closed 2026-11-08 by JamBalaya56562