motioneye-project/motioneye #2025: MJPEG Stream Artifacts

chauser10, 2021-02-07:
I am trying to get a stream from a pi zero w running [pikrellcam](https://github.com/billw2/pikrellcam) in the motion eye os addon in home assistant. I have been able to add other network camera's just fine. I am able to view the stream from the browser and vlc without issue. An help would be greatly appreciated!

https://user-images.githubusercontent.com/22107538/107157926-24248d80-6944-11eb-9404-3c847471ce1f.mp4

starbasessd, 2021-02-07:
Have you tried the PiZero (assuming you're using a CSI camera) with MotionEyeOS set to Fast Network Camera?

chauser10, 2021-02-07:
I had been using MotionEyeOS for the past month or so with success. I was just giving pikrellcam a try because I have heard it has good performance and local motion detection.

JamBalaya56562, 2026-11-08:
Closing, as this got no further replies, and it was with the Home Assistant add-on of 2021. The stream played fine in the browser and VLC, so the artifacts came from Motion reading it as a network camera. Please open a new issue with `motion.log` if a pikrellcam stream still shows artifacts with a current release.

closed 2026-11-08 by JamBalaya56562