motioneye-project/motioneye #2026: Accessing interface when installed on Rasbian

Ru-Pirie, 2021-02-08:
I have installed the software onto a Raspberry Pi following the guide with no errors.
I couldn't work out how to access the interface. I tried the IP of the Pi along with its hostname, I tried http https nothing seemed to work any ideas?

[image]

starbasessd, 2021-02-08:
http://[ip_addres]:8765
if using motionEye on anything but motionEyeOS. (RPiOS, Debian, Ubuntu,
etc...

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Ru-Pirie, 2021-02-08:
Your not gona believe this... I forgot to restart the service after install.

closed 2021-02-08 by Ru-Pirie

starbasessd, 2021-02-08:
Pretty common... Oops. Enjoy...

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?