motioneye-project/motioneye #2027: API Access

sdevo619, 2021-02-08:
So I am trying to find some examples of api calls for the motion eye api. I have already made it pretty far down the rabbit hole here, but I am confused on a couple things here. If I want to perform a GET to retrieve data from the motion eye camera, how is the Authorization portion supposed to be formatted, and can you provide an example with dummy data. I am familiar with how this typically works, but I keep getting 404 errors trying to access the API. I have been through the API Docs and seen the information provided there, and the GET request is definitely making it to the camera and back, but seems to be getting stuck due to authorization. Are there any settings that needs to changed to allow API access on the Motion Eye Cam Software? 

Does the Authorization need to be in the HTTPHeaderField or not??

Also, I know it says that the API is a draft and not implemented, so I just wanted to make sure the API is actually working as it seems that post is rather old, and the documentation, suggests it is working at this point. Thank you!!

starbasessd, 2021-02-08:
Flagging @ccrisan

ccrisan, 2021-02-09:
The API hasn't been implemented yet and there are no plans to implement it any time soon. Sorry about the bummer.

JamBalaya56562, 2026-11-10:
Closing as a duplicate of #972, which tracks an API for other systems; as ccrisan said, the API draft in the wiki was never implemented, so there is no authorization format to use for it.

closed 2026-11-10 by JamBalaya56562