motioneye-project/motioneye #2029: Samba Foscam

thom13, 2021-02-10:
Hi, 

I have installed motioneye on docker, and I want to save image with samba, it works with my 2 raspberry pi cam with motioneye installed on it. But samba is not available for my foscam... Foscam is on Network Camera type. Before this installation I had motioneye installed on Home assistant and Samba was available for my Fosacm. What is the reason ?

starbasessd, 2021-02-10:
HA installs SMB/CIFS support when it installs (IIRC) and is there under the
motionEye.
You can run the sudo apt install smbclient in your base OS
If you give me your setup, Which PI and which base OS, I will try to get
you  some setup instructions in the morning, US EST...

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

thom13, 2021-04-02:
Thanks, but Samba is available for the other cams in motioneye, just this one is not available, the foscam camera...

starbasessd, 2021-04-02:
If you give me your setup, Which PI and which base OS, I will try to get
you  some setup instructions in the morning, US EST
Samba is not an option in motion/motionEye, unless it is installed in the BaseOS. If you are using motionEyeOS you have the option to act both as a server and connect to Samba servers to save files.

starbasessd, 2021-04-02:
Reconfirmed.
Installed RaspberrPiOS on 32GB drive
Installed into Pi3B+
Booted
ssh'd into Pi
Update & Upgrade PiOS
Set up localization, password, etc
Install docker, add user Pi to group docker
reboot
Run docker command
docker run --name="motioneye" \
    -p 8765:8765 \
    --hostname="motioneye" \
    -v /etc/localtime:/etc/localtime:ro \
    -v /etc/motioneye:/etc/motioneye \
    -v /var/lib/motioneye:/var/lib/motioneye \
    --restart="always" \
    --detach=true \
    ccrisan/motioneye:master-armhf
After install connect to http://ip_address:8765
The docker image doesn't have Samba/CIFS installed in the docker.
Installing any of the Install motion/motioneye in XXX flavor of linux doesn't have support for Samba/CIFS either.
Samba/CIFS IS built into motionEyeOS.

JamBalaya56562, 2026-11-08:
Closing with the reason: the storage options of a camera come from the motionEye server that runs it. The two Pi cameras are run by their own motionEye, where network shares are on, while the Foscam is run by the motionEye in Docker, which offers Network Share only with `smb_shares true` in its `motioneye.conf`, `cifs-utils` installed, and the right to mount, which a container does not have by default. Mounting the share on the host and passing it into the container as a volume, then choosing it as a custom path, avoids all of that.

closed 2026-11-08 by JamBalaya56562