motioneye-project/motioneye #2031: Home Assistant on a hyper-v with motioneye - port 8081-8084 not open

Stupoo27, 2021-02-11:
Hello,

I have been trying for hours and cant figure out how to open ports 8081:8084 on motion eye so i can use the streaming options.

So far i have figured out after installing from add-ons it seems to run in a docker container however i cant figure out how to change the docker so its got those ports open. Seems i need to edit the docker run command to include those ports?

Can anyone advise how i go about doing that or know of another way? Apologies i have little experience with this setup.

Thanks

starbasessd, 2021-02-11:
I believe there is a file you set the ports you need in the HA wrapper created by FRENCK, you may want to ask over at their forum:
https://community.home-assistant.io/t/home-assistant-community-add-on-motioneye/71826

Stupoo27, 2021-02-13:
Thanks. Found a config (motioneye.conf) file with a mention or tcp listening ports but it's got a port I don't recognise so don't think that's what I'm looking for. Asked over in the home assistant group but no replies. The embedded links work but they don't seem to work when trying to add to the camera streams in home assistant. If I run a command to see the running docker containers it's showing blank for the ports.

starbasessd, 2021-02-13:
It's in the yaml for docker run of whatever HA uses for starting motionEye,
but I don't know so I can't really help you...

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Stupoo27, 2021-02-15:
Being doing some more digging and found the docker-compose.yml files for motioneye but they already seem to have the ports needed in there. If i look at the ports listening on the server i can actually see the default motion eye ones listening and when i enable port forwarding the port im now testing with 8082 from the internet to the home assistant server IP but its still not working. My firewall is off completely on the windows 10 device running the Hyper-V. I ran a local scan for open ports on the homeassistant server and the correct ports are indeed open (8081,8082,8083,8084) so looks like the problem must be with motioneye i think.

starbasessd, 2021-02-15:
I run motion/motionEye in a docker, with 5 cameras, with no issue. My
docker install is:
docker run --name="motioneye" -p 8765:8765 -p 8081:8081 -p 8082:8082 -p
8083:8083 -p 8084:8084 -p 8085:8085 --hostname="motioneye" -v
C:\VM\DOCKER_config\motioneye:/etc/motioneye -v
C:\VM\DOCKER_config\motioneye:/var/lib/motioneye --restart="always"
--detach=true  ccrisan/motioneye:master-amd64
(on a single line) and works fine.
Also works with a Win10 host, Ubuntu host, and RaspberryPiOS host. None
have an issue with any of the 8081-8085 ports.
The only difference is the HA interface. Have you gotten ANY answers from
the Community or FRENCK?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Stupoo27, 2021-02-15:
Seems its randomly started to work! Perfectly as well. Hard to say what was the problem as i dont think i changed anything apart from trying to activate web-control to "true" but that's not going anything to do with the video streaming. Strange. Thanks for your replies

JamBalaya56562, 2026-11-08:
Closing, as it started working for you, as your last reply says. The streaming ports are published by the Home Assistant add-on, which is maintained by the Home Assistant community add-ons project.

closed 2026-11-08 by JamBalaya56562