motioneye-project/motioneye #2032: Unable to play recorded movies in browser, no thumbnails, missing first keyframe

AndreVallestero, 2021-02-12:
I am unable to play recorded movies in the browser even though I can download each movie and play it on my phone or desktop. Additionally, all movie thumbnails are missing, and ~30% of the time, the first keyframe is missing for a movie.

[image]

[image]

These issues persist regardless of the device, OS, and browser.

- Raspberry Pi OS Lite
- Motion 4.2.2 (Issue persists in Motion 4.3.1)
- Latest Motioneye
- Raspberry Pi 4 4GB
- h264 1280x960 @20fps

starbasessd, 2021-02-12:
Tried h264-omx?
Using Movie Passthrough?
Reduce Quality to something under 100% (75-90%)?
Reduce framerate to 10FPS?
Which camera, how connected (wifi or ethernet) and RTSP or HTTP?

AndreVallestero, 2021-02-12:
> Tried h264-omx?

I know what OMX is, but I'm not sure how to enable it with motion[eye]. Do I need to manually compile libffmpeg with OMX enabled?

> Using Movie Passthrough?

Yes

> Reduce Quality to something under 100% (75-90%)?

Tried qualities 50%, 75%, 90%, issues persist

> Reduce framerate to 10FPS?

Tried framerate 10, 15, 10, issues persist

> Which camera, how connected (wifi or ethernet) and RTSP or HTTP?

Generic Hangzhou Xiongmai / XM camera with RTSP / ONVIF over PoE / Ethernet (I'm aware of the SoC vulnerability, the device is airgapped from the rest of my networked)

starbasessd, 2021-02-12:
None of the things I mentioned would affect existing recordings, only new
ones.
What instructions are you using to install motion/motionEye?
If you turn off Movie Passthrough, you get multiple options for format of
video to save.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

AndreVallestero, 2021-02-12:
I made sure to reboot and see if the new movies had the same issues. They still do.

Here are my install commands:
```
wget https://github.com/Motion-Project/motion/releases/download/release-4.2.2/pi_buster_motion_4.2.2-1_armhf.deb
sudo dpkg -i pi_buster_motion*
sudo pip install motioneye
sudo mkdir -p /etc/motioneye /var/lib/motioneye
sudo cp /usr/local/share/motioneye/extra/motioneye.conf.sample /etc/motioneye/motioneye.conf
sudo cp /usr/local/share/motioneye/extra/motioneye.systemd-unit-local /etc/systemd/system/motioneye.service
sudo systemctl daemon-reload
sudo systemctl enable motioneye
sudo systemctl start motioneye
```

starbasessd, 2021-02-12:
I would try the instructions from here:
https://github.com/ccrisan/motioneye/wiki/%28Install-On-Ubuntu-%2820.04-or-Newer%29
They work on any Debian based system (RaspberryPiOS, Ubuntu, Debian, etc...)
I use it regularly to install on VMs.
The biggest issue has been the upstream removal of support for Python2.7
and pip2, so these instructions install both.
Be SURE to run each line with 'sudo' or switch to root using sudo su or
sudo -i

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

AndreVallestero, 2021-02-12:
Python 2.7 is the default for Raspberry Pi OS so I haven't run into dependency issues. I followed all the commands (or their equivalent for Raspberry Pi OS) but the issue still persists.

starbasessd, 2021-02-12:
Run the command:
pip --version
and post the results

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

AndreVallestero, 2021-02-12:
pip 18.1 from /usr/lib/python2.7/dist-packages/pip (python 2.7)

starbasessd, 2021-02-12:
Fresh image of RaspberryPiOS (1/11/2021) Lite.
sudo apt update && sudo apt upgrade -y
reboot
sudo -i
apt-get install ssh curl motion ffmpeg v4l-utils -y
apt-get install python2 -y
curl https://bootstrap.pypa.io/2.7/get-pip.py --output get-pip.py
python2 get-pip.py
apt-get install libffi-dev libzbar-dev libzbar0 -y
apt-get install python2-dev libssl-dev libcurl4-openssl-dev libjpeg-dev -y
apt-get install python-pil -y
pip2 install motioneye
mkdir -p /etc/motioneye
cp /usr/local/share/motioneye/extra/motioneye.conf.sample /etc/motioneye/motioneye.conf
mkdir -p /var/lib/motioneye
cp /usr/local/share/motioneye/extra/motioneye.systemd-unit-local /etc/systemd/system/motioneye.service
systemctl daemon-reload
systemctl enable motioneye
systemctl start motioneye
pip2 install motioneye --upgrade
systemctl restart motioneye
[image]
[image]
Working as expected.

AndreVallestero, 2021-02-12:
> apt-get install ssh curl motion ffmpeg

I was missing `ffmpeg` that solved my issues, I thought we only needed `libffmpeg` my bad 😅. Thank you so much for taking the time to help me.

closed 2021-02-12 by AndreVallestero