motioneye-project/motioneye #2034: Action Button area is locked to a static width; hides more than 8 buttons

fouram, 2021-02-13:
I have a camera set up that has pan/tilt/zoom functionality, as well as presets. If I set up up/down/left/right/zoom_in/zoom_out and some presets (I currently have 4 total), plus the snapshot button makes 11 total. However, only 8 show up.

I've traced this back to 14b2f52 in main.js, where the `width` attribute for div.camera-action-buttons-wrapper is set to a static 10em on line 4154:

`cameraOverlay.find('div.camera-action-buttons-wrapper').css('width', 4 * 2.5 + 'em');`

Directly above this line, commented out, is this:

`//cameraOverlay.find('div.camera-action-buttons-wrapper').css('width', Math.ceil(cameraConfig.actions.length / 2) * 2.5 + 'em');`

...which, with 11 buttons, will calculate the width to 15em, which is not only enough to display them all, but also confines them neatly to two rows. Is there a reason that the dynamic sizing was removed? It is obscuring many of the buttons (the presets, in my case) and unless I'm missing something, the alternative seems to work just fine. Perhaps something done for testing and accidentally left in the commit? :)

I'd submit a pull for this, but I'm not sure i'm properly set up to do so.

johanblaise, 2021-08-25:
Hello, 
it seems i've a similar problem.
Only two presets are displayed instead of 9.

`[root@meye-58427a3b etc]# ls preset
preset1_1  preset2_1  preset3_1  preset4_1  preset5_1  preset6_1  preset7_1  preset8_1  preset9_1  `

JamBalaya56562, 2026-10-30:
The fixed width of the action button area was removed by @sevimuelli in #2903 (released with 0.43.1): the buttons now wrap into two rows. I checked on the current `dev` with 17 buttons (pan/tilt, zoom and 9 presets): all are shown, at desktop and phone widths. Closing as completed.

closed 2026-10-30 by JamBalaya56562