motioneye-project/motioneye #2036: 4th added Cam shows preview of the 1st cam

FrankyMaker, 2021-02-14:
Hi,

if i add the 4th Cam, it shows always the preview of the 1st added cam.
Also it records the 1st cam.

If i add the cam again (while the 4th is inside) as the 5th, then its correct.
After that i had to remove the 4th.

regards
Frank

zagrim, 2021-02-15:
That can happen if the streaming port of the 4th cam is the same as the streaming port of the first cam. What I can't tell is why that would happen (as it seems to happen to some users sometimes).

JamBalaya56562, 2026-11-10:
Closing, as this got no follow-up to zagrim's answer: a camera showing another camera's picture has the same streaming port as that one, so giving the 4th camera a free port under Video Streaming fixes it. Please open a new issue if a new camera still gets a port that is in use with a current release.

closed 2026-11-10 by JamBalaya56562