motioneye-project/motioneye #2037: Freeze OrangePi MotionEye

thalesmaoa, 2021-02-15:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: motionEye Version 0.41 / Motion 4.2.2+gitUNKNOWN.

### Board Model

I am using the following board/model: Orange Pi One.

### Camera

I am using the following type of camera: Cheap Chinese Brand (V380 - Compatible).

My camera model is: Generic.

### Network Connection

My motionEyeOS unit is connected to the network via: Ethernet Cable.

### Peripherals

I am using the following peripherals that I consider relevant to this issue: None

### Log Files

I consider the following log files relevant to this issue:

motioneye.log
messages.log
 
Hi there, re-opening the issue in the correct spot (#https://github.com/ccrisan/motioneyeos/issues/2670).
I never faced a single problem since I start using MotionEye. However, I'm facing it now. Since I'm using an Orange Pi, I was convinced that it was the problem. However, after looking the logs, I'm not so sure.

1) Recently my local network provider changed my modem and they obligated me to have IPv4 + IPv6. I can't disable IPV6. MotionEyeOS worked without a problem since I manually set IPv4. However, after another freeze and looking into messages.log, I found the following:

```
Feb 15 03:19:53 motioneyeOS user.err motion: [3:ml3:Porta] [ERR] [ALL] motion_init: Error capturing first image
Feb 15 04:18:12 motioneyeOS user.notice network: eth0 has no IP address
Dec 31 21:00:08 motioneyeOS daemon.err rpcbind[1211]: cannot create socket for udp6
Dec 31 21:00:08 motioneyeOS daemon.err rpcbind[1211]: cannot create socket for tcp6
Dec 31 21:00:08 motioneyeOS user.info sntp[1340]: sntp 4.2.8p13@1.3847-o Wed Nov  6 17:33:54 UTC 2019 (1)
Dec 31 21:00:09 motioneyeOS user.info sntp[1340]: 1969-12-31 21:00:03.078170 (+0300) +1613396224.529016 +/- 1075597483.054597 0.pool.ntp.org 198.27.76.102 s2 no-leap
```

Am I wrong or it just lost its IP? It is also trying to get an IPv6? How do I completely disable it?

2) A also found another interesting log inside motioneye.log

```
2021-02-15 10:40:27: [motioneye]     INFO: downloading firmware version 20200606
2021-02-15 10:40:35: [motioneye]     INFO: extracting firmware
2021-02-15 10:41:05: [motioneye]    ERROR: Exception in callback <functools.partial object at 0xb40a6330>
Traceback (most recent call last):
  File "usr/lib/python2.7/site-packages/tornado/ioloop.py", line 758, in _run_callback
  File "usr/lib/python2.7/site-packages/tornado/stack_context.py", line 300, in null_wrapper
  File "/usr/lib/python2.7/site-packages/motioneye/platformupdate.py", line 54, in perform_update
Exception: firmware extracting failed
```

It is not updating? Is there a way to fix it?

[motioneye.log](https://github.com/ccrisan/motioneyeos/files/5982029/motioneye.log)
[messages.log](https://github.com/ccrisan/motioneyeos/files/5982035/messages.log)

[motioneye.log](https://github.com/ccrisan/motioneye/files/5983446/motioneye.log)
[messages.log](https://github.com/ccrisan/motioneye/files/5983449/messages.log)

thalesmaoa, 2021-02-19:
Since I have my motioneye freezed again I tried to fix it. I have deleted some files and updated my motioneye using cli. 

After that, my motioneye keeps power cycling. Please, check the boot log at the image bellow.
Strange is that its saying that many files are read-only. Is there a way to enable write access?

[image]

starbasessd, 2021-02-19:
I'm pretty sure the / file system is trashed after the attempted upgrade.
The file system  is normally read only for most of the tree to protect it.
The normal way to enable read write is to use the Debug feature under
Expert Settings in the WebGUI
Your easiest fix at this point is to re-image.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

thalesmaoa, 2021-02-20:
I did as you mentioned. I had a image from last year and re-flashed. However, I'm not able to upgrade.

```
# fwupdate extract
extracting...
losetup: /data/.fwupdate/firmware.img: No such file or directory
[root@motioneyeOS .fwupdate]# ll /data/.fwupdate/firmware.img
-rw-------    1 root     root      330.0M Feb 20 16:37 /data/.fwupdate/firmware.img
``` 

However, its not creating `extract_done` file. Is there a way to debug?

starbasessd, 2021-02-20:
https://github.com/ccrisan/motioneyeos/wiki/FAQ#software-updates
That version may not be able to do an automatic update. There were a number
of updates (especially on Pi Hardware, that weren't upgradable due to
several issues, the only way to upgrade was to do a new image.
If you need to go that route, I'd do a backup, first, then restore after
the new image is installed.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

thalesmaoa, 2021-02-21:
I did build my system with pip. Can I try to upgrade using pip?

JamBalaya56562, 2026-11-08:
Closing, as this is about motionEyeOS 0.41 on an Orange Pi One, which is maintained in its own repository (motioneye-project/motioneyeos). The read-only files and the failed update are part of motionEyeOS, and as starbasessd said, a new image with a restored backup is the way out.

closed 2026-11-08 by JamBalaya56562