motioneye-project/motioneye #2038: Unable to scale streams to smaller size

sduffs, 2021-02-19:
Good afternoon and thank you for this project. I have motioneye running in a docker container on Ububntu 20.04. At the localhost browser I am able to view all of my four camera streams but only in a single vertical layout. I have set the preference to vertical layout with four layout rows but there are no changes. It may be important to note as well, that changing any setting in the U.I. underneath the preferences section, does not prompt me to apply the new settings. I've attached screenshots of my U.I, json.config, motion.config and motioneye.config. I apologize if I haven't approached this issue in the wrong manner. I was computer illiterate until early 2020. Thank you for your time.

 
[image]
[image]
[image]
[image]
[image]
[image]

starbasessd, 2021-02-19:
Normally, there is another option In Preferences, above the "Fit Frames
Vertically.
What browser are you using, on the Ubuntu 20.04?
Is it on a PC or Pi/SBC?
Do you have another PC to check from?
Which instructions did you follow to install the docker container, and
which docker are you running?
I will spin up an Ubuntu 20.04 here in a few minutes to test.
(BTW, thanks for the screen shots...)

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

sduffs, 2021-02-19:
It's running on an old gateway pc connected through a serial connection. 
I get the same results on chrome and firefox. 

Accessing Motioneye over the network from my laptop gives me the additional option of Layout Columns, as well as the desired multi-stream display. 

I would attach a screenshot of it working properly, but that will require tutorial time to learn how to transfer files from a windows to a linux machine. 

I will try from another pc with a serial connection, as well as from the same PC using the display port with the same monitor. 

Would it be helpful to the community to share my results on this thread?

Thank you for getting back to me in so quickly. This is a (now obvious) step in troubleshooting that I hadn't considered. You've helped me beyond my motioneye issue.

starbasessd, 2021-02-19:
Interesting.
Yes it's something that others might benefit from.
Ubuntu VM still spinning up. (I have grandkids I watch before and after
their school)
It could be seeing the connection across Serial as a 'mobile'. In your
browsers, look to see in Settings if it offers to View Desktop Site' option
on the machine(s) that have issues...

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2021-02-19:
Getting Hash incorrect errors trying to do apt update / apt upgrade on most
servers, so it may have to wait a while (hours) until I can update &
upgrade & install on the VM... Sorry.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2021-02-20:
Clean install of Ubuntu 20.04
Update && upgrade
Install docker.io
run:
docker run --name="motioneye" \
    -p 8765:8765 \
    --hostname="motioneye" \
    -v /etc/localtime:/etc/localtime:ro \
    -v /etc/motioneye:/etc/motioneye \
    -v /var/lib/motioneye:/var/lib/motioneye \
    --restart="always" \
    --detach=true \
    ccrisan/motioneye:master-amd64
Using Firefox (default browser on Ubuntu 20.04)
http://localhost:8765
Connection shows correct for Layout Columns as first option

JamBalaya56562, 2026-11-08:
Closing, as this was with the web UI of 0.42.1 (2021), and the same setup showed Layout Columns from your laptop. Today Layout Columns is always shown under Preferences, and until it is chosen, the columns follow the window width, about one per 500 px. Preferences are saved in the browser as soon as they change, so they need no Apply. Please open a new issue if the layout still doesn't change with a current release.

closed 2026-11-08 by JamBalaya56562