motioneye-project/motioneye #2039: "Run a Command" ; "Run a End Command" Question Help Wanted

lnxdread, 2021-02-21:
I have motioneye running on a Rasp 3b+
I tried to run a _Python_ script whenever a motion is detected.

/home/pi/motioneye-stuff/motion_info.py

This does not work.

Is this even possible with the "Run a Command" option?

starbasessd, 2021-02-21:
You may have to prefix your path with
python, python2, or python3 depending on which version of python it was
written in.
Also, instead, you could edit the python script to add as line #1
#!/usr/bin/env python
and set the executable bit
sudo chmod +x /home/pi/motioneye-stuff/motion_info.py

<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virus-free.
www.avg.com
<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

lnxdread, 2021-02-22:
You Sir solved my problem.
Make the Script **executable** and the correct prefix do the job.
`python /home/pie/motioneye-stuff/motion_info.py`

Thanks for your help

closed 2021-02-22 by lnxdread