motioneye-project/motioneye #2041: USB Webcam - Distortions, picture in picture overlay

curatortux, 2021-02-28:
Hi all,

issue preview:
https://i.ibb.co/HCH1Gq2/Video-21-02-28-07-42-27.gif

thought my USB webcam was defective so i bought a new one.
The same issue still occurs. Funny thing is, that these distortions sometimes(?) get recognized as motion even.
motionEye Version 0.42.1
Motion Version 4.3.1
Have found a lot of "artifacts" or "pixel distortion" threads so far, but nothing that compares to these. 
Out of my depth and hoping for good ideas. Thanks all!

starbasessd, 2021-02-28:
Which SBC? 
What BaseOS?

curatortux, 2021-02-28:
Sorry i don't know what SBC stands for.
BaseOS is motioneyeOS which is based on https://buildroot.org/

starbasessd, 2021-02-28:
SBC = Single Board Computer (Raspberry PI and equivalents)
Which version of motionEyeOS?
Which Pi?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

JamBalaya56562, 2026-11-08:
Closing, as this got no answer to which board and motionEyeOS version it was, and it was with motionEye 0.42.1 and Motion 4.3.1. The picture of a USB camera comes from Motion, so the same distortion with two cameras points to the board, its USB power, or Motion's capture settings. Please open a new issue with `motion.log` if a USB camera still shows this with a current release.

closed 2026-11-08 by JamBalaya56562