motioneye-project/motioneye #2042: When will the privacy mask on motioneye debain 10

titof2375, 2021-02-28:
hello, I am on debian 10 (Linux 5.4.78-2-pve), motion is in version (4.3.2) and motioneye (0.42.1), my system is on proxmox in 6.3.3.
I installed montion without problem on the other hand I don't have the masck privacy and when I install it on an rpi 4 I have it, there is a way to have it in the amd64 version

starbasessd, 2021-02-28:
Which version of motionEye/motionEyeOS on the Pi4?

titof2375, 2021-03-03:
the which version 4.3.2 and 0.421 sur rpi 4

starbasessd, 2021-03-03:
Spinning up a Debian 10.8.0 VM.
From the motion-project site, mask_privacy is available in 4.1 and newer.
Privacy Mask is available in motionEye 0.42.1 in motionEyeOS
I do not have the option in motionEye 0.42.1, motion 4.2.2, on Ubuntu 20.04
I will test & report.

starbasessd, 2021-03-03:
Confirmed issue in Ubuntu 20.04, motionEye 0.42.1, motion 4.2.2, directly editing camera-1.conf or using extra motion options fields.

starbasessd, 2021-03-03:
Confirmed issue in Debian 10.8 Buster, motionEye 0.42.1, motion 4.2.2, and 4.3.2

titof2375, 2021-03-04:
hello, so you confirm that with debian 10.8 and Ubuntu 20.04, there is no privacy mask option

starbasessd, 2021-03-04:
Correct. Also confirmed in RaspberryPiOS, motion 4.3.2 and motionEye 0.42.1, doesn't exist.
Only exists / works in motionEyeOS.

zagrim, 2021-03-05:
Privacy mask support didn't make it to ME 0.42.1 release (the changes were merged on Jun 29, 2020, 0.42.1 was freezed around Jun 6, 2020), but unless the maintainer decides to revert that altogether, it will be in the next release (whenever that may come). And as far as the timeline goes, even in MEOS it should only be available in dev20201026 and not in any "official" releases, right?
But anyway, I see no reason why those changed wouldn't eventually be part of next releases.

starbasessd, 2021-03-05:
@zagrim @ccrisan The weird thing is the version # for motionEye wasn't updated between 20200606 & dev20201026, which confused even me... Probably just overlooked, in the grand scheme of things. Hopefully it will be rectified for the next release. It IS a 'very nice'(tm) feature.

starbasessd, 2021-03-05:
'bug' label removed.

zagrim, 2021-03-06:
I don't really know how the MEOS releases and pre-releases are being made, especially regarding what is being copied over from ME repository, but I guess the "dev" pre-releases might be using ME "dev" branch as their source for ME code, something which is sometimes called a "development snapshot". But this is just guessing. Yeah, it might be good to clarify that also in the bundled ME version string.

zagrim, 2021-03-06:
Kind of related to this is that I've couple of times hoped that there would be some sort of roadmap (even a vague one) on what is planned to be included in the next ME release, if it will be just smaller features and bugfixes or some more major (like Python 3 support), and most importantly some sort of timeline. But I guess with ccrisan clearly lacking time or energy for ME related things lately and the amount of support from community contributors being rather erratic, that's plenty to be wishing for.

So, all there is for now is that whatever features there are that have been merged so far (their pull request having been merged, more specifically), but are not mentioned in the changelog of the latest release (https://github.com/ccrisan/motioneye/releases), will eventually be available in the next release somewhere in the future.

JamBalaya56562, 2026-11-08:
Closing this, as the privacy mask is part of motionEye since 0.43.1, the first release after 0.42.1: as zagrim explained, it was merged just after 0.42.1 was frozen. With motionEye 0.43.1 or newer and Motion 4.1 or newer, it is under Video Device > Privacy Mask on Debian as well.

closed 2026-11-08 by JamBalaya56562